swiss-seismological-service/scdetect #40: compilation error using seiscomp4.6.0

Donavin97, 2021-07-07:
Hello everyone.
I am getting a strange error when trying to compile scdetect using seiscomp4.6.0 on ubuntu20.04.2LTS.
I have installed all required dependencies and currently using cmake 3.16.
The error I am receiving is as follows.
"
-- Adding pkg DETECT
-- ... resides in /home/seismo/scsrc/src/extras/scdetect.
-- ... will be installed under /home/seismo/seiscomp.
CMake Error at src/extras/scdetect/src/apps/scdetect/CMakeLists.txt:34 (if):
  if given arguments:

    "EQUAL" "DEBUG"

  Unknown arguments specified

-- Configuring incomplete, errors occurred!
See also "/home/seismo/scsrc/build/CMakeFiles/CMakeOutput.log".
See also "/home/seismo/scsrc/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:5: all] Error 1
seismo@SeisComP:~/scsrc$ gg
"
Could someone assist?
Thanks.

damb, 2021-07-07:
Note that the module still must be considered as WIP and is highly experimental.

Anyway, did you set the [CMAKE_BUILD_TYPE](https://cmake.org/cmake/help/latest/variable/CMAKE_BUILD_TYPE.html) when generating the project buildsystem? If set to `Debug` `scdetect` is complied with [Address Sanitizer](https://github.com/google/sanitizers/wiki/AddressSanitizer) enabled. However, in case the `CMAKE_BUILD_TYPE` is left empty it should be handled correctly, too.

Donavin97, 2021-07-07:
Hi there,
yes, I now set the build type to Debug, it generated the build correctly, I am almost done compiling the sources.
Will let you know what happens when the process is over.
Also, thanks for making such an awesome tool.
I am running a SeisComP system with a sparce network, as a result, I miss some events due to it not being detected on 4 or more stations.
However, I think scdetect will solve quite a few problems for me.
I'll follow you, so that I can test your future projects.
I'd also like to contribute, in future, findings, suggestions, etc.
Thanks again.
Regards: Donavin.

Donavin97, 2021-07-07:
Hi again,
The compilation works now.
However, at 98% I ran into this error:
"/home/seismo/scsrc/src/extras/scdetect/src/apps/scdetect/eventstore.cpp:54:12: error: ‘databaseArchive’ was not declared in this scope; did you mean ‘setDatabaseArchive’?
   54 |       po = databaseArchive()
      |            ^~~~~~~~~~~~~~~
      |            setDatabaseArchive
/home/seismo/scsrc/src/extras/scdetect/src/apps/scdetect/eventstore.cpp:59:5: error: ‘setCached’ was not declared in this scope
   59 |     setCached(cached);
      |     ^~~~~~~~~
make[2]: *** [src/extras/scdetect/src/apps/scdetect/CMakeFiles/scdetect.dir/build.make:193: src/extras/scdetect/src/apps/scdetect/CMakeFiles/scdetect.dir/eventstore.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:11072: src/extras/scdetect/src/apps/scdetect/CMakeFiles/scdetect.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
seismo@SeisComP:~/scsrc/build$ 
"
How can I solve this?
Thanks.

damb, 2021-07-07:
This is due to a recent change I made upstream at [SeisComP/common](https://github.com/SeisComP/common/commit/9b6c3184148720b66ae5582eca4e68d78c933308). It should work once you install [SeisComP v4.6.1](https://github.com/SeisComP/seiscomp/releases/tag/4.6.1).

damb, 2021-07-07:
> I'd also like to contribute, in future, findings, suggestions, etc.

Thanks. That sounds very good. If there issues, let me know.

Note also that at the time being there is no stable official release. This is mainly due to the lack of some basic features which are still missing (e.g. magnitude computation, etc.). However, we are working on this.

Donavin97, 2021-07-08:
Hi,
Thanks, the compilation work, now just for the testing to begin.
I'm going to close this now, but I will test the module extensively, I want to see how it performs against South African earthquakes.
Thanks and regards.

closed 2021-07-08 by Donavin97