swiss-seismological-service/scdetect #41: error when trying to run scdetect on own data

Donavin97, 2021-07-08:
Good day,
I have successfully installed and configured scdetect.
When running the test files all works well.
However, I am using mysql as my main database, and when trying to run scdetect using my own data I get an error.

seismo@SeisComP:~$ seiscomp exec scdetect --templates-json seiscomp/etc/scdetect/templates.json --inventory-db seiscomp/etc/scdetect/inv.scml --event-db seiscomp/etc/scdetect/event.scml -I seiscomp/etc/scdetect/d.mseed --ep=scd.scml --debug
15:54:49 [debug] Adding plugin path: .
15:54:49 [debug] Adding plugin path: /home/seismo/.seiscomp/plugins
15:54:49 [debug] Adding plugin path: /home/seismo/seiscomp/lib/plugins
15:54:49 [debug] Adding plugin path: /home/seismo/seiscomp/lib
15:54:49 [debug] Adding plugin path: /home/seismo/seiscomp/share/plugins
15:54:49 [debug] Adding plugin path: /home/seismo/seiscomp/share/plugins/scdetect
15:54:49 [debug] Trying to open plugin at /home/seismo/seiscomp/share/plugins/dbmysql.so
15:54:49 [info] Plugin dbmysql registered
15:54:49 [debug] Trying to open plugin at /home/seismo/seiscomp/share/plugins/hypo71.so
15:54:49 [info] Plugin hypo71 registered
15:54:49 [debug] Trying to open plugin at /home/seismo/seiscomp/share/plugins/locnll.so
15:54:49 [info] Plugin locnll registered
15:54:49 [debug] Trying to open plugin at /home/seismo/seiscomp/share/plugins/locext.so
15:54:49 [info] Plugin locext registered
15:54:49 [debug] Trying to open plugin at /home/seismo/seiscomp/share/plugins/md.so
15:54:49 [info] Plugin md registered
15:54:49 [debug] Trying to open plugin at /home/seismo/seiscomp/share/plugins/dbsqlite3.so
15:54:49 [info] Plugin dbsqlite3 registered
15:54:49 [info] 
Plugins:
--------
 [1]
  description: MySQL database driver
       author: GFZ Potsdam <seiscomp-devel@gfz-potsdam.de>
      version: 0.9.2
          API: 15.0.0
 [2]
  description: HYPO71 seismic hypocenter locator plugin
       author: IPGP <www.ipgp.fr>
      version: 0.2.0
          API: 15.0.0
 [3]
  description: Locator implementation using NonLinLoc by Anthony Lomax (http://alomax.free.fr/nlloc/)
       author: Jan Becker, gempa GmbH
      version: 0.7.2
          API: 15.0.0
 [4]
  description: Locator wrapper for external scripts
       author: gempa GmbH <seiscomp-devel@gempa.de>
      version: 0.1.0
          API: 15.0.0
 [5]
  description: Md duration magnitude plugin
       author: IPGP <www.ipgp.fr>
      version: 0.1.2
          API: 15.0.0
 [6]
  description: SQLite3 database driver
       author: GFZ Potsdam <seiscomp-devel@gfz-potsdam.de>
      version: 0.9.0
          API: 15.0.0

15:54:49 [info] Connect to messaging
15:54:49 [debug] Trying to connect to scdetect@localhost/production with primary group = LOCATION
15:54:49 [debug] Attempt to connect to localhost:18180/production
15:54:49 [debug] Registered clientname: scdetect
15:54:49 [info] Outgoing messages are encoded to match schema version 0.11
15:54:49 [info] Want to flush 0 backlog messages
15:54:49 [info] Loading complete inventory from seiscomp/etc/scdetect/inv.scml
15:54:49 [info] Finished loading complete inventory
15:54:49 [debug] reading boundary polygons from file: /home/seismo/seiscomp/share/fep/reg.fep
15:54:49 [debug] reading boundary polygons from file: /home/seismo/seiscomp/share/fep/regions.fep
15:54:49 [debug] Number of PolyRegions loaded: 2
15:54:49 [debug] Total number of vertices read in: 8
15:54:50 [info] Found cities.xml and read 71653 entries
15:54:50 [debug] trying to open stream file://seiscomp/etc/scdetect/d.mseed
15:54:50 [info] Loading events from seiscomp/etc/scdetect/event.scml
15:54:50 [debug] Found database version v0.11
Segmentation fault (core dumped)
seismo@SeisComP:~$ 

I'll attach some files for you to look at.
Attached is an archive with my miniseed event data file, event params, inventory and template files.
Also what's funny, is that I can't seem to figure out how the event Id's work, on my seiscomp it just states: "Origin/ID"
However, when I use the fdsnws service and retrieve that same event in quakeml format, the origin ID's are: "smi:org.gfz-potsdam/geofon/Origin/ID".
Can anyone give me advice what to do?
Any help would be appreciated.
Thanks and regards: Donavin.
[scd.tar.gz](https://github.com/damb/scdetect/files/6785057/scd.tar.gz)

damb, 2021-07-09:
Thanks a lot for reporting.

There seem to be a few issues with regards to the input data you provided:
- If the catalog is provided as a file it must be `SCML` formatted - not `QuakeML` (AFAIK SeisComP does not provide facilities which allow reading from `QuakeML` directly.)
- The `templates.json` is mal-formatted. It must be a JSON array of detector configurations i.e. you're missing the top-level `[]`. I see that this might not have been clear from the documentation. I hope it is clearer now.
- Besides, there seems to be no waveform data available in order to create the template waveform. See the logs below:
```
damb@seiscomp-fedora32 ~/tmp/iss-41 $ scdetect --offline --playback --templates-json templates.json --inventory-db inventory.scml --event-db catalog.scml -I data.mseed --ep=detections.scml --debug
11:04:22 [error/Application] DEPRECATION WARNING: You are using the parameter 'recordstream.service' or 'recordstream.source' which will be removed in the next major release. Please remove the old parameters and set the new 'recordstream' parameter to 'slink://localhost:18000' instead.
11:04:22 [debug] Adding plugin path: .
11:04:22 [debug] Adding plugin path: /home/damb/.seiscomp/plugins
11:04:22 [debug] Adding plugin path: /home/damb/seiscomp/lib/plugins
11:04:22 [debug] Adding plugin path: /home/damb/seiscomp/lib
11:04:22 [debug] Adding plugin path: /home/damb/seiscomp/share/plugins
11:04:22 [debug] Adding plugin path: /home/damb/seiscomp/share/plugins/scdetect
11:04:22 [debug] Trying to open plugin at /home/damb/seiscomp/share/plugins/dbpostgresql.so
11:04:22 [info] Plugin dbpostgresql registered
11:04:22 [info] 
Plugins:
--------
 [1]
  description: PostgreSQL database driver
       author: GFZ Potsdam <seiscomp-devel@gfz-potsdam.de>
      version: 0.11.0
          API: 15.0.0

11:04:22 [info] Loading complete inventory from inventory.scml
11:04:23 [info] Finished loading complete inventory
11:04:23 [debug] trying to open stream file://data.mseed
11:04:23 [info] Playback mode enabled
11:04:23 [info] Loading events from catalog.scml
11:04:23 [debug] Found database version v0.11
11:04:23 [debug] Found database version v0.11
11:04:23 [info] Finished loading events
11:04:23 [info] Loading template configuration from /home/damb/tmp/iss-41/seiscomp/etc/scdetect/templates.json
11:04:23 [debug] Creating detector processor (id=detector-01) ... 
11:04:23 [debug] GT.BOSA.10.SHZ (GT.BOSA.10.SHZ): Using arrival pick: origin=smi:org.gfz-potsdam.de/geofon/Origin/20210701101921.29327.8103, time=2021-07-01T01:39:48.847856Z, phase=Pg, stream=GT.BOSA.10.SHZ
11:04:23 [debug] GT.BOSA.10.SHZ (GT.BOSA.10.SHZ): Loaded stream from inventory for epoch: start=2021-07-01T01:39:46.847856Z, end=2021-07-01T01:39:50.847856Z
11:04:23 [debug] Creating template processor (id=template-02) ... 
11:04:23 [debug] trying to open stream file://data.mseed
11:04:23 [warning] GT.BOSA.10.SHZ (GT.BOSA.10.SHZ): No waveform data for template processor available. Skipping.
11:04:23 [debug] GT.LBTB.10.SHZ (GT.LBTB.10.SHZ): Using arrival pick: origin=smi:org.gfz-potsdam.de/geofon/Origin/20210701101921.29327.8103, time=2021-07-01T01:39:38.094293Z, phase=Pg, stream=GT.LBTB.10.SHZ
11:04:23 [debug] GT.LBTB.10.SHZ (GT.LBTB.10.SHZ): Loaded stream from inventory for epoch: start=2021-07-01T01:39:36.094293Z, end=2021-07-01T01:39:40.094293Z
11:04:23 [debug] Creating template processor (id=template-01) ... 
11:04:23 [debug] trying to open stream file://data.mseed
11:04:23 [warning] GT.LBTB.10.SHZ (GT.LBTB.10.SHZ): No waveform data for template processor available. Skipping.
11:04:23 [debug] Object removed from cache with id: 73
11:04:23 [debug] Object removed from cache with id: 74
11:04:23 [debug] Object removed from cache with id: 75
11:04:23 [debug] This message will not be repeated for other objects
11:04:23 [debug] Application initialized
11:04:23 [info] Starting record acquisition
11:04:24 [info] Finished acquisition
11:04:24 [info] Sending close event after finishing acquisition
11:04:24 [info] Close event received, returning
11:04:24 [debug] Found 0 origins.
11:04:24 [debug] Leaving ::done
11:04:24 [info] Waiting for record thread
```

- Could you please check the changes I implemented in #42?
- Please, also reread the docs and check if it is clearer now.
- Besides it would be good if you'd provide data in order to create the template waveforms. The MiniSEED you provided covers only the following traces (w.r.t. the stream identifiers specified in the template configuration):
```
GT.BOSA.10.SHZ | 2021-07-06T04:49:52.350000Z - 2021-07-06T05:00:08.250000Z | 40.0 Hz, 24637 samples
GT.LBTB.10.SHZ | 2021-07-06T04:49:56.600000Z - 2021-07-06T05:00:03.175000Z | 40.0 Hz, 24264 samples
```

PS: In future, would you be so kind and use Markdown formatting for your reports. Thanks.

damb, 2021-08-11:
@Donavin97, any updates regarding this?

Donavin97, 2021-09-10:
Hi @damb, I haven't tested your latest version yet, but will do so ASAP.
I do have a question though.
When trying to detect small events using a single stream, the earthquake location is allways in the same location as the master event, is this the way it should be?
I thought that one can use scdetect to detect seismic events with there own location propperties, independent from the master event.
I hope I am making sence?
Thanks.

damb, 2021-09-13:
> When trying to detect small events using a single stream, the earthquake location is always in the same location as the master event, is this the way it should be?
> I thought that one can use scdetect to detect seismic events with there own location properties, independent from the master event.

Yes. This is the way it should be (at least at the time being).

PR #42 (2021-11-16): Improve error handling
**Changes**:
- Avoid SegmentationFault in case of reading the catalog from a file failed.
- Improve log messages
- Clarify templates configuration formatting

Closes: #41
closed 2021-11-16 by damb