swiss-seismological-service/scdetect #54: sc3ml output file version not compatible with obspy

mmesim, 2021-11-02:
The output file after running scdetect (for example results.scml) is in version 0.12. 

When I try to read it with obspy

`catalog=read_events('results.scml',format="SC3ML")`

I get the following error: 

`ValueError: Can't read SC3ML version 0.12, ObsPy can deal with versions [0.5, 0.6, 0.7, 0.8, 0.9, 0.10].`

What is the best solution? Make the output obspy-compatible in the code or find a work around and convert the output to an obspy-compatible version afterwards?

damb, 2021-11-02:
`scdetect` relies on SeisComP facilities when exporting event parameters to SC3ML. I didn't dig into the details when it comes to enforcing a certain version to be written (if it is possible at all). Though, there is a possible workaround: Convert your `result.scml` to `QUAKEML` using the XSLT from https://raw.githubusercontent.com/SeisComP/common/master/libs/xml/0.12/sc3ml_0.12__quakeml_1.2.xsl. AFAIU, obspy is able to read QuakeML 1.2.

mmesim, 2021-11-03:
Maybe it is more of an obspy issue, I don't disagree.

Note: 
I followed the process you described and the converted quakeml.xml was empty. 

However, I tried this 

https://github.com/SeisComP/common/blob/9c6056945a0ebb8abc57ce67f7fdac57ebbf7238/libs/xml/0.12/sc3ml_0.12__quakeml_1.2-RT.xsl

and then

` xsltproc -o quakeml.xml sc3ml_0.12__quakeml_1.2.xsl result.scml`

Still I cannot read the quakeml.xml output.

damb, 2021-11-03:
> Anyway, this issue must be considered as an obspy issue rather than a SeisComP/scdetect issue.

Actually, this is not correct. Instead, I didn't get it right when it comes to the purpose of the actual data formats:

- QUAKEML is a data format for a fully processed complete catalog. I.e. an `Origin` not being associated to an `Event` is simply not taken into consideration. (For this reason, you receive an empty `quakeml.xml` after processing the SC3ML with the XSLT).
- QUAKEML-RT is more flexible i.e. it may include `Origin`s without being associated to an `Event`.
- SC3ML is SeisComP specific. It is similar to QUAKEML-RT and is designed to be used within a real-time processing environment.

Though, AFAIU, obspy's `read_event` is not designed to work with *unassociated* origins (an [obspy catalog](https://docs.obspy.org/packages/autogen/obspy.core.event.Catalog.html) is a container for `Event` objects). Besides, it isn't possible to import QUAKEML-RT. Therefore, if you'd like to use `scdetect`'s output within obspy, the only way I see is to wrap the origins into *pseudo* events with e.g. `scevent`. Note that this may require tweaking `scevent` related configuration parameters in order to obtain the desired pseudo events.

Wrapping into pseudo events with `scevent`:
```
$ ~/seiscomp/bin/seiscomp exec scevent --debug --disable-db --ep result.xml > catalog.xml
```
The `catalog.xml` output is SC3ML formatted.

Next, convert the data to QUAKEML (alternatively, as a quick and dirty solution, simply change the SC3ML version number in the document in order to satisfy the obspy requirements):
```
$ xsltproc --verbose -o quakeml.xml --stringparam ID_PREFIX smi:org.gfz-potsdam.de/geofon/ sc3ml_0.12__quakeml_1.2.xsl catalog.xml
```

Now, you should be able to read the *intermediate* `scdetect` output (i.e. your `result.xml`) with obspy:
```
$ python3 -c "from obspy import read_events; cat = read_events('quakeml.xml'); print(cat)"
```

I hope this helps.

Thanks @fmassin for all the details.

mmesim, 2021-11-03:
Yes, it works! 

> Note that this may require tweaking scevent related configuration parameters in order to obtain the desired pseudo events.

Yes, I had to set **minimumDefiningPhases** to 1, as I'm working with one stream. 

Thanks @damb !

closed 2021-11-03 by damb