swiss-seismological-service/scdetect #57: Multi-stream attempt returns zero detections

mmesim, 2021-11-10:
I'm not sure if I should edit something on the json file but it seems that multi-stream detection doesn't return any detections. 

- Attempt 01: ACC | One network code [CH]

json file
[{
  "detectorId": "detector-01",
  "createArrivals": true,
  "createTemplateArrivals": false,
  "gapInterpolation": true,
  "gapThreshold": 0.1,
  "gapTolerance": 1.5,
  "triggerDuration": 0.5,
  "triggerOnThreshold": 0.5,
  "triggerOffThreshold": 0.0,
  "originId": "smi:ch.ethz.sed/sc3a/origin/NLL.20191105125505.255283.1897990",
  "filter": "BW_BP(2,1.5,15)",
  "templateFilter": "BW_BP(2,1.5,15)",
  "streams": [
    {
      "templateId": "template-01",
      "initTime": 10,
      "templateWaveformStart": -0.5,
      "templateWaveformEnd": 2,
      "waveformId": "CH.SAYF2..HGZ",
      "templateWaveformId": "CH.SAYF2..HGZ",
      "templatePhase": "Pg"
   },
   {
      "templateId": "template-02",
      "initTime": 10,
      "templateWaveformStart": -0.5,
      "templateWaveformEnd": 2,
      "waveformId": "CH.STSW2..HGZ",
      "templateWaveformId": "CH.STSW2..HGZ",
      "templatePhase": "Pg"
   }
 ]
}]

Command:

`scdetect --offline --playback --templates-json templates/templates.json  --inventory-db stations/CH.xml --event-db events/myevents.xml -I data/data.two.streams.CH.mseed --ep=results.scml --amplitudes-force=0  --debug `

**Note**: the mseed file contains both streams. 

> 11:59:40 [debug] Adding plugin path: .
> 11:59:40 [debug] Adding plugin path: /home/maria/.seiscomp/plugins
> 11:59:40 [debug] Adding plugin path: /home/maria/seiscomp/lib/plugins
> 11:59:40 [debug] Adding plugin path: /home/maria/seiscomp/lib
> 11:59:40 [debug] Adding plugin path: /home/maria/seiscomp/share/plugins
> 11:59:40 [debug] Adding plugin path: /home/maria/seiscomp/share/plugins/scdetect
> 11:59:40 [debug] Trying to open plugin at /home/maria/seiscomp/share/plugins/dbpostgresql.so
> 11:59:41 [info] Plugin dbpostgresql registered
> 11:59:41 [info] 
> Plugins:
> 
>  [1]
>   description: PostgreSQL database driver
>        author: GFZ Potsdam <seiscomp-devel@gfz-potsdam.de>
>       version: 0.11.0
>           API: 15.0.0
> 
> 11:59:41 [info] Loading complete inventory from stations/CH.xml
> 11:59:42 [info] Finished loading complete inventory
> 11:59:42 [debug] trying to open stream file://data/data.two.streams.CH.mseed
> 11:59:42 [info] Playback mode enabled
> 11:59:42 [info] Loading events from events/myevents.xml
> 11:59:42 [debug] Found database version v0.12
> 11:59:42 [debug] Found database version v0.12
[... 760 lines cut ...]
      "templateWaveformId": "8D.RAW2..HHZ",
      "templatePhase": "Pg"
   },
   {
      "templateId": "template-02",
      "initTime": 10,
      "templateWaveformStart": -0.5,
      "templateWaveformEnd": 2,
      "waveformId": "CH.SENIN..HHZ",
      "templateWaveformId": "CH.SENIN..HHZ",
      "templatePhase": "Pg"
   }
 ]
}]  

Command: 

` scdetect --offline --playback --templates-json templates/templates.json  --inventory-db stations/stations.scml --event-db events/myevents.xml -I data/data.two.streams.mseed --ep=results.scml --amplitudes-force=0  --debug
`

**Note:** 
stations.scml  = contains both networks
data.two.streams.mseed = contains both streams

Thoughts?

damb, 2021-11-10:
Thanks for the report.

For debugging. Could you please upload the data you used? I.e.
- inventory (`station.scml`)
- catalog (`myevents.xml`)
- waveform data

Thanks.

mmesim, 2021-11-10:
[files.zip](https://github.com/damb/scdetect/files/7512792/files.zip)

damb, 2021-11-10:
Apparently, the waveform data is not *prepared* correctly (i.e. records must be sorted by endtime). Simply prepare the data with e.g.

```
$ ~/seiscomp/bin/seiscomp exec scmssort -u -E data.two.streams.CH.mseed > data.sorted.CH.mseed
```

SeisComP's [waveform playback tutorial](https://www.seiscomp.de/doc/base/tutorials/waveformplayback.html) describes in detail how to run playbacks and how to prepare the waveform data, accordingly.

mmesim, 2021-11-11:
Thanks @damb. It works now. 
Probably we should update the documentation, somewhere under this

[Waveform data and RecordStream configuration](https://github.com/damb/scdetect#waveform-data-and-recordstream-configuration)

damb, 2021-11-12:
@mmesim, this is a good point. Thanks.

I'll close this here. With #58 I created a follow-up.

closed 2021-11-12 by damb