swiss-seismological-service/scdetect #66: Multi-station detection

mmesim, 2021-12-06:
This issue is similar to [#62](https://github.com/damb/scdetect/issues/62). 

Here I attempted to run a multi-station detection. Same problems as in multi-channel detection. Results do not match the detections performed on single station. 
 
[18.test.zip](https://github.com/damb/scdetect/files/7659775/18.test.zip)

damb, 2021-12-06:
Thanks, @mmesim. I'll have a look into it and come back.

damb, 2021-12-21:
Note that the behaviour from #62 was caused by a bug in the implementation which should be fixed with #72, now. Therefore, when performing your checks, again, please make also make use of the `"arrivalOffsetThreshold"` detector configuration option. This is particularly relevant for a multi-station setup.

mmesim, 2022-02-04:
I found the same problem as in #62. In more detail: 

Test-04 : RAW2.HHZ | Ndetections : **88** | 04.txt [catalog with origin times & CC] 
[04.test.txt](https://github.com/damb/scdetect/files/8004351/04.test.txt)

```json
[{
  "detectorId": "detector-01",
  "createArrivals": true,
  "createTemplateArrivals": false,
  "gapInterpolation": true,
  "gapThreshold": 0.1,
  "gapTolerance": 1.5,
  "triggerDuration": 0.5,
  "triggerOnThreshold": 0.5,
  "triggerOffThreshold": 0.3,
  "originId": "smi:ch.ethz.sed/sc3a/origin/NLL.20191105125505.255283.1897990",
  "filter": "BW_BP(2,1.5,15)",
  "templateFilter": "BW_BP(2,1.5,15)",
  "streams": [
    {
      "templateId": "template-HHZ",
      "initTime": 10,
      "templateWaveformStart": -0.5,
      "templateWaveformEnd": 2,
      "waveformId": "8D.RAW2..HHZ",
      "templateWaveformId": "8D.RAW2..HHZ",
      "templatePhase": "Pg"
   }
 ]
}]
```
Test-09 : SENIN.HHZ | Ndetections : **34** | 09.txt [catalog with origin times & CC] 
[09.test.txt](https://github.com/damb/scdetect/files/8004372/09.test.txt)

```json
[{
  "detectorId": "detector-01",
  "createArrivals": true,
  "createTemplateArrivals": false,
  "gapInterpolation": true,
  "gapThreshold": 0.1,
  "gapTolerance": 1.5,
  "triggerDuration": 0.5,
  "triggerOnThreshold": 0.5,
  "triggerOffThreshold": 0.3,
  "originId": "smi:ch.ethz.sed/sc3a/origin/NLL.20191105125505.255283.1897990",
  "filter": "BW_BP(2,1.5,15)",
  "templateFilter": "BW_BP(2,1.5,15)",
  "streams": [
    {
      "templateId": "template-HHZ",
      "initTime": 10,
      "templateWaveformStart": -0.5,
      "templateWaveformEnd": 2,
      "waveformId": "CH.SENIN..HHZ",
      "templateWaveformId": "CH.SENIN..HHZ",
      "templatePhase": "Pg"
   }
 ]
}]
```

Test-18 : RAW2.HHZ & SENIN.HHZ | Ndetections : **26** | 18.txt [catalog with origin times & CC] 
[18.test.txt](https://github.com/damb/scdetect/files/8004380/18.test.txt)

```json
{
  "detectorId": "detector-01",
  "createArrivals": true,
  "createTemplateArrivals": false,
  "gapInterpolation": true,
  "gapThreshold": 0.1,
  "gapTolerance": 1.5,
  "triggerDuration": 0.5,
  "triggerOnThreshold": 0.5,
  "triggerOffThreshold": 0.0,
  "arrivalOffsetThreshold":1.0,
  "originId": "smi:ch.ethz.sed/sc3a/origin/NLL.20191105125505.255283.1897990",
  "filter": "BW_BP(2,1.5,15)",
  "templateFilter": "BW_BP(2,1.5,15)",
  "streams": [
    {
      "templateId": "template-01",
      "initTime": 10,
      "templateWaveformStart": -0.5,
      "templateWaveformEnd": 2,
      "waveformId": "8D.RAW2..HHZ",
      "templateWaveformId": "8D.RAW2..HHZ",
      "templatePhase": "Pg"
   },
   {
      "templateId": "template-02",
      "initTime": 10,
      "templateWaveformStart": -0.5,
      "templateWaveformEnd": 2,
      "waveformId": "CH.SENIN..HHZ",
      "templateWaveformId": "CH.SENIN..HHZ",
      "templatePhase": "Pg"
   }
 ]
}]
```

"expected results" based on the common events after single-station detection: **17** events
[expected.results.txt](https://github.com/damb/scdetect/files/8004440/expected.results.txt)

I also attach the scml files and data. 

[data.zip](https://github.com/damb/scdetect/files/8004444/data.zip)
[scml.zip](https://github.com/damb/scdetect/files/8004446/scml.zip)

damb, 2022-02-04:
@mmesim, just for clarification. Did you have already a closer look into the code? Did you debug the issue with a debugger, already?

mmesim, 2022-02-04:
No. No.

mmesim, 2022-02-09:
I tested again the detections on the two stations. 

Test-04 : RAW2.HHZ | Ndetections : 173 (originally it detected 88)
Test-09 : SENIN.HHZ | Ndetections : 66 detections (originally it detected 34)

Test-18 : RAW2.HHZ & SENIN.HHZ | Ndetections : 56 [all true detections, waveforms checked]

The results agree with the common detections performed on single stations. 

@damb Well done!