swiss-seismological-service/scdetect #68: Amplitude-Magnitude regression attept

mmesim, 2021-12-13:
This might not be an issue but I definitely need some help, despite the very detailed documentation. 

I run the following command

`scdetect --offline --playback --templates-json templates/templates.json  --inventory-db stations/stations.scml --event-db events/myevents.xml -I data/data.sorted.two.streams.mseed --ep=results.scml --amplitudes-force=0 --templates-family-json templates/templateFamilyConfiguration.json --debug`

I get the following 

> 15:47:44 [info] Loading template configuration from /home/maria/mydata/templates/templates.json
15:47:44 [debug] Creating detector processor (id=detector-01) ... 
15:47:44 [debug] [8D.RAW2..HHZ (8D.RAW2..HHZ)] using arrival pick: origin=smi:ch.ethz.sed/sc3a/origin/NLL.20191105125505.255283.1897990, time=2019-11-05T04:23:49.099929Z, phase=Pg, stream=8D.RAW2..HHZ
15:47:44 [debug] [8D.RAW2..HHZ (8D.RAW2..HHZ)] loaded stream from inventory for epoch: start=2019-11-05T04:23:48.59993Z, end=2019-11-05T04:23:51.099929Z
15:47:44 [debug] [8D.RAW2..HHZ (8D.RAW2..HHZ)] creating template waveform processor with id: template-HHZ
15:47:44 [debug] [detector-01::template-HHZ] [8D.RAW2..HHZ (8D.RAW2..HHZ)] filters configured: filter="BW_BP(2,1.5,15)"
15:47:44 [info] Loading template family configuration from /home/maria/mydata/templates/templates.json
15:47:44 [debug] [e7bc36fc-95e4-45de-a33d-6dcef43ca809] Creating template family (id=e7bc36fc-95e4-45de-a33d-6dcef43ca809) ...
terminate called after throwing an instance of 'Seiscomp::detect::builder::NoBindings'
  what():  [8D.RAW2.] failed to load bindings configuration
Aborted (core dumped)

I think I configured the bindings correctly in scconfig. Then save and run `seiscomp update-config` . 

Here is the template and the templateFamily Configuration

> [{
  "detectorId": "detector-01",
  "createArrivals": true,
  "createTemplateArrivals": false,
  "gapInterpolation": true,
  "gapThreshold": 0.1,
  "gapTolerance": 1.5,
  "triggerDuration": 0.5,
  "triggerOnThreshold": 0.5,
  "triggerOffThreshold": 0.3,
  "originId": "smi:ch.ethz.sed/sc3a/origin/NLL.20191105125505.255283.1897990",
  "filter": "BW_BP(2,1.5,15)",
  "templateFilter": "BW_BP(2,1.5,15)",
  "streams": [
    {
      "templateId": "template-HHZ",
      "initTime": 10,
      "templateWaveformStart": -0.5,
      "templateWaveformEnd": 2,
      "waveformId": "8D.RAW2..HHZ",
      "templateWaveformId": "8D.RAW2..HHZ",
      "templatePhase": "Pg"
   }
 ]
}]

>  [
  {
    "id": "e7bc36fc-95e4-45de-a33d-6dcef43ca809",
    "references": [
      {
        "detectorId": "detector-01",
        "streams": [
          {
            "templateWaveformId": "8D.RAW2..HHZ",
            "upperLimit": 2
          }
        ]
      }
     ]
    }
   ]

Thanks.

damb, 2021-12-13:
You need to specify where to obtain the bindings configuration from:
- Make sure, you assigned the bindings to the corresponding station(s) (e.g. `8D.RAW2`). (Key files may be generated easily with [`scconfig`](https://www.seiscomp.de/doc/apps/scconfig.html).)
- With `seiscomp update-config` you store the bindings configuration within your SeisComP DB (e.g. PostgreSQL, etc.). With `--offline` you disable messaging and thus you need to explicitly pass `-d|--database URL` on the CLI.
- Alternatively, use the `--config-db URL` to specify on the CLI where to obtain the bindings configuration from. This can be a file, in case you previously dumped to bindings configuration by means of e.g. [`scxmldump`](https://www.seiscomp.de/doc/apps/scxmldump.html). E.g.
```
scxmldump -fC -o config.scml -d postgres://user:pass@localhost:5432/seiscomp"
```

Afterwards you run `scdetect` with
```
scdetect --config-db "file://$(realpath config.scml)" ...
```

Dumping the configuration might be helpful, too, when you want to verify the bindings configuration is available in the database.

damb, 2021-12-13:
Anyway, `Seiscomp::detect::builder::NoBindings` should be handled properly. I'm going to fix this.

damb, 2021-12-14:
> Anyway, `Seiscomp::detect::builder::NoBindings` should be handled properly. I'm going to fix this.

Fixed with #69.

mmesim, 2021-12-14:
Now there is a warning 

> 10:56:01 [warning] [8D.RAW2.] failed to load bindings configuration. Skipping template family initialization.

but `scdetect` continues to run without calculating magnitudes. Still trying to figure out why it doesn't find the bindings even though I added 

`--config-db bindings/config.scml`

where I get `config.scml` from

`--config-db bindings/config.scml`

damb, 2021-12-14:
> Update: the `config.scml` seems to be empty. Ugh!

There it is. Make sure, you correctly assigned the bindings to the station.

mmesim, 2021-12-14:
`bindings2cfg --key-dir ./etc/key -o config.scml`

I was giving the wrong path to the **etc/key**. In my case should be $HOME/seiscomp/etc/key. 

Could you update this in the documentation?

damb, 2021-12-14:
> Could you update this in the documentation?

Done with 5b1e8ab13719945aae16de3a70570a8051f36b59.

Note that this particular example I've taken from the [bindings2cfg documentation](https://www.seiscomp.de/doc/apps/bindings2cfg.html) straight away. Also, the documentation is explicitly referring to the [bindings2cfg documentation](https://www.seiscomp.de/doc/apps/bindings2cfg.html)  for further information and details.

mmesim, 2021-12-14:
> Done with 5b1e8ab.

Note that this particular example I've taken from the bindings2cfg documentation straight away. Also, the documentation is explicitly referring to the bindings2cfg documentation for further information and details.

Thanks

mmesim, 2021-12-14:
So, it seems that I did a few things wrong. First, the bindings file, which after using the right path it works fine. 

It also seems that I was trying to compute magnitudes on the vertical component (HHZ) which is not feasible with this method. 

Now I'm trying to use the two horizontals from the RAW2 station. 

I get this warning 

> 11:24:58 [warning] [e7bc36fc-95e4-45de-a33d-6dcef43ca809] Missing template family members. Skipping template family registration.

and my `templateFamilyConfiguration.json` looks like 

> [
  {
    "id": "e7bc36fc-95e4-45de-a33d-6dcef43ca809",
    "references": [
      {
        "detectorId": "detector-01",
        "streams": [
          {
            "templateWaveformId": "8D.RAW2..HHE",
            "upperLimit": 2
          },
          {
            "templateWaveformId": "8D.RAW2..HHN",
            "upperLimit": 2
          }
        ]
      }
     ]
    }
   ]

any clue?

mmesim, 2021-12-14:
Also this might be helpful

```11:45:30 [debug] [detector-01::template-01] Initialize stream: samplingFrequency=200.000000
11:45:30 [debug] [detector-01] Detector result (triggering) (2019-11-05T04:23:51.71Z - 2019-11-05T04:23:51.71Z): fit=1.000000, arrival_count=2
11:45:30 [warning] [detector-01] 8D.RAW2..HHE: Failed to prepare result. Reason: failed to lookup offset from POT. Resetting.
```

damb, 2021-12-14:
> It also seems that I was trying to compute magnitudes on the vertical component (HHZ) which is not feasible with this method.
> 
> Now I'm trying to use the two horizontals from the RAW2 station.

Independently, how the detector is configured, `scdetect` will try compute the amplitudes for the horizontal components regarding a sensor location. This requires that
- bindings configuration for the horizontal components must be available
- inventory data (i.e. sensor response related data) for the horizontal components must be available
- and of course waveform data must be available, too.

damb, 2021-12-14:
@mmesim, it's difficult to guess, what's going wrong, here. Could you please share your data

- `inventory.scml`
- `catalog.scml` (including station magnitudes)
- `config.scml` (bindings configuration)
- `templates.json`
- `templates-family.json`
- `data.mseed` (waveform data including the waveform data both required for detection and required for amplitude calculation)

mmesim, 2021-12-14:
Sure. Here it is. 

[20.test.zip](https://github.com/damb/scdetect/files/7711023/20.test.zip)

damb, 2021-12-14:
@mmesim, the `catalog.scml` you provided does not include station magnitudes for the corresponding events. This is the reason, why template family registration failed and the message

> 11:24:58 [warning] [e7bc36fc-95e4-45de-a33d-6dcef43ca809] Missing template family members. Skipping template family registration.

is logged.

damb, 2021-12-14:
> Also this might be helpful
>
> ```
> 11:45:30 [debug] [detector-01::template-01] Initialize stream: samplingFrequency=200.000000
>
> 11:45:30 [debug] [detector-01] Detector result (triggering) [...]

Apparently, this seems to be an unrelated issue. I'll have a closer look into it and come back.

mmesim, 2021-12-14:
> NOTE: At the time being, you cannot use fdsnws-event to dump a catalog including station magnitudes. You may use scxmldump, instead (of course, this requires to have access to the DB backend).

I see. I'll ask for access to DB to extract that info. For now I cannot do much. 

Thanks a lot. 

We can leave the issue open until I run successfully a test and get magnitudes. Do you agree?

damb, 2021-12-14:
> Apparently, this seems to be an unrelated issue. I'll have a closer look into it and come back.

Ok. I've got it. See #71. Thanks for the report @mmesim.

damb, 2021-12-14:
> We can leave the issue open until I run successfully a test and get magnitudes. Do you agree?

Fine for me.

damb, 2022-02-15:
`scdetect-cc` computes so called *StationMagnitudes* (i.e. actually *SensorLocationMagnitudes*). In the [SeisComP datamodel](https://www.seiscomp.de/doc/base/api-python.html#) StationMagnitudes are child objects of an *Origin*.

mmesim, 2022-02-15:
Ok, thanks for this. I still do not get magnitudes. 

```
scdetect-cc --offline --playback --templates-json templates/templates.json --inventory-db stations/stations.scml --event-db events/template.scml -I data/data.sorted.RAW2.horizontals.onehour.mseed --ep=results.scml  --config-db bindings/config.scml  --templates-family-json templates/templateFamilyConfiguration.json --debug
```
- First I didn't update the template configuration. Now it contains 
``` json
 "createMagnitudes":true,
 "createAmplitudes":true,
```
- The `template.scml`  contains StationMagnitudes. I extract the file from the db. Also I checked  that the detections are ok with this file. 
- So perhaps the problem is with the bindings? I do not know. Are global bindings required or only the `scdetect` ones?

damb, 2022-02-15:
Does the `template.scml` contain amplitudes?

From the docs:
> Amplitude-magnitude regression based magnitude type. Besides, of the corresponding amplitudes to be computed, this particular type requires both amplitudes and station magnitudes to be available by means of [event parameters](https://github.com/damb/scdetect#inventory-events-and-configuration).

damb, 2022-02-15:
Besides:
- Does your catalog contain StationMagnitudes of type `MLh|MLhc`?
- Does your inventory contain the metadata for the horizontal instrument components? Recall, that the regression is based on the horizontal components, only.

mmesim, 2022-02-15:
>Does the template.scml contain amplitudes?

> From the docs:
>
> Amplitude-magnitude regression based magnitude type. Besides, of the corresponding amplitudes to be computed, this particular type requires both amplitudes and station magnitudes to [...]

Just fixed that.

mmesim, 2022-02-15:
> Besides:
> 
> * Does your catalog contain StationMagnitudes of type `MLh|MLhc`?
> * Does your inventory contain the metadata for the horizontal instrument components? Recall, that the regression is based on the horizontal components, only.

-MLh

-Yes

mmesim, 2022-02-15:
I think the problem is the sh*ity  templateFamilyConfiguration.json

mmesim, 2022-02-15:
I updated the json file but still cannot get magnitudes. 

``` json
[
  {
    "id": "e7bc36fc-95e4-45de-a33d-6dcef43ca809",
    "references": [
      {
        "detectorId": "detector-01",
        "streams": [
          {
            "templateWaveformId": "8D.RAW2..HHE",
            "upperLimit": 2
          },
          {
            "templateWaveformId": "8D.RAW2..HHN",
            "upperLimit": 2
          },
          {
           "originId": "smi:ch.ethz.sed/sc3a/origin/NLL.20191105125505.255283.1897990",
            "streams": [
          {
            "templatePhase": "Sg",
            "templateWaveformId": "8D.RAW2..HHN",
            "templateWaveformStart": -0.5,
            "templateWaveformEnd": 2
          },
          {
            "templatePhase": "Sg",
            "templateWaveformId": "8D.RAW2..HHE",
            "templateWaveformStart": -0.5,
            "templateWaveformEnd": 2
          }
        ]
      }
     ]
    }
   ]
  }
]

damb, 2022-02-15:
Well, the formatting of the `templates-family.json` is simply not correct. Again, from the docs:
> `"references"`: Required. A JSON array of template family members configuration objects. Members may reference detectors (defined by the [template configuration](https://github.com/damb/scdetect#template-configuration)) or third-party origins (i.e. origins which are not used for detections but contribute to the amplitude-magnitude regression).

Also, it should be sufficient to reference a stream referring to a sensor location just once.

damb, 2022-02-15:
@mmesim, recall that the binary was renamed to `scdetect-cc`. However, `8D.RAW2` has the deprecated `scdetect` bindings assigned.

mmesim, 2022-02-15:
> @mmesim, recall that the binary was renamed to scdetect-cc. However, 8D.RAW2 has the deprecated scdetect bindings assigned.

Ok, that's dump @damp!!! I should have noticed that.

damb, 2022-02-15:
> I do not think that is intuitive at all to understand the json configuration.

Improvements are always welcome, @mmesim. Go ahead!

Note also, that the configuration by means of the JSON configuration file is kind of a *temporary solution*. In future, the configuration might be loaded from e.g. a database, alternatively.

mmesim, 2022-02-15:
Do you think it will work with one template?

damb, 2022-02-15:
Of course. Though, the more samples there are the better the regression.

mmesim, 2022-02-15:
Does this seem right?

``` json

[
  {
    "id": "e7bc36fc-95e4-45de-a33d-6dcef43ca809",
    "references": [
      {
        "detectorId": "detector-01",
        "streams": [
          {
            "templateWaveformId": "8D.RAW2..HHE",
            "upperLimit": 10
          },
          {
            "templateWaveformId": "8D.RAW2..HHN",
            "upperLimit": 10
          }
        ]
      },
      {
        "originId": "smi:ch.ethz.sed/sc3a/origin/NLL.20191105125505.255283.1897990",
        "streams": [
          {
            "templatePhase": "Sg",
            "templateWaveformId": "8D.RAW2..HHE",
            "templateWaveformStart": -0.5,
            "templateWaveformEnd": 2
          }
        ]
      }
    ]
  }
]

damb, 2022-02-15:
The `templates-family.json` looks better now.

> I get this error

This should happen :bug:. In order to debug, I need your configuration. Alternatively, you execute `scdetect-cc` under `gdb` and post the stacktrace ...

mmesim, 2022-02-15:
> How does your `templates.json` look like?

``` json

  "filter": "BW_BP(2,1.5,15)",
  "templateFilter": "BW_BP(2,1.5,15)",
  "streams": [
    {
      "templateId": "template-01",
      "initTime": 10,
      "templateWaveformStart": -0.5,
      "templateWaveformEnd": 2,
      "waveformId": "8D.RAW2..HHE",
      "templateWaveformId": "8D.RAW2..HHE",
      "templatePhase": "Sg"
   },
   {
      "templateId": "template-02",
      "initTime": 10,
      "templateWaveformStart": -0.5,
      "templateWaveformEnd": 2,
      "waveformId": "8D.RAW2..HHN",
      "templateWaveformId": "8D.RAW2..HHN",
      "templatePhase": "Sg"
   }
 ]
}]

damb, 2022-02-15:
@mmesim, within a template family configuration there is no need to reference the origin (i.e. `"smi:ch.ethz.sed/sc3a/origin/NLL.20191105125505.255283.1897990"`) twice. It is already resolved by means of the first member (i.e. referenced by means of `"detectorId": "detector-01"`). Therefore, your `templates-family.json` should look like:

```json
[
  {
    "id": "e7bc36fc-95e4-45de-a33d-6dcef43ca809",
    "references": [
      {
        "detectorId": "detector-01",
        "streams": [
          {
            "templateWaveformId": "8D.RAW2..HHE",
            "upperLimit": 10
          }
        ]
      }
    ]
  }
]
```
Also, it is enough to reference a sensor location just once (i.e. `8D.RAW2..HH`).

mmesim, 2022-02-16:
Good morning! #90 fixed the problem! We have magnitudes!! :raised_hands: :tada:

damb, 2022-02-16:
Good morning!

@mmesim, a few notes from my side:
- With #84 the waveform buffer size is configurable. If buffering waveform data is disabled (or rather the waveform buffer size is too small), no amplitudes (and thus no magnitudes) will be computed.
- Besides, I noted that the `MLx` amplitude values are pretty low. Maybe, you can look into this issue?
- After all these efforts here it would be good to write it somewhere down. What about providing a tutorial? E.g. under https://github.com/damb/scdetect/wiki. Since the `README.md` is pretty bloated anyway, it would be good to *modularize* the documentation, as well. What do you think? Do you volunteer?

mmesim, 2022-02-16:
> Besides, I noted that the MLx amplitude values are pretty low. Maybe, you can look into this issue?

This is the next step. I wanted to make sure it works and then we will check the results in detail! Keep in mind that the earthquake I use for detection is already small and it is normal to detect smaller events. If these events were larger they would have been detected in the routine analysis. This doesn't change the fact that we should look at the results very carefully! 

> After all these efforts here it would be good to write it somewhere down. What about providing a tutorial? E.g. under https://github.com/damb/scdetect/wiki. Since the README.md is pretty bloated anyway, it would be good to modularize the documentation, as well. What do you think? Do you volunteer?

This is a great suggestion! Of course I'm in. It would be nice to clean the README.md in order not to scare people. I agree to go to the wiki and have separate sections. I already have plenty of notes and it would be share! 

**Last thing before Closing the issue.** 

I run again the `scdetect-cc` with this template family configuration

``` json
[
  {
    "id": "e7bc36fc-95e4-45de-a33d-6dcef43ca809",
    "references": [
      {
        "detectorId": "detector-01",
        "streams": [
          {
            "templateWaveformId": "8D.RAW2..HHE",
            "upperLimit": 10
          }
        ]
      },
      {
	"originId": "smi:ch.ethz.sed/sc3a/origin/NLL.20191105133225.294125.1901273",
        "streams": [
	{
        "templatePhase": "Sg",
	"templateWaveformId": "8D.RAW2..HHE",
	"templateWaveformStart": "-0.5",
	"templateWaveformEnd": "2"
	}
	]	
      },
       {
        "originId": "smi:ch.ethz.sed/sc3a/origin/NLL.20191105132334.75776.1900431",
        "streams": [
        {
        "templatePhase": "Sg",
        "templateWaveformId": "8D.RAW2..HHE",
        "templateWaveformStart": "-0.5",
        "templateWaveformEnd": "2"
        }
        ]
      },
            {
        "originId": "smi:ch.ethz.sed/sc3a/origin/NLL.20191105131428.857539.1899532",
        "streams": [
        {
        "templatePhase": "Sg",
        "templateWaveformId": "8D.RAW2..HHE",
        "templateWaveformStart": "-0.5",
        "templateWaveformEnd": "2"
        }
        ]
      }
    ]
  }
]
```
I checked one event and the results are slightly different, which makes sense (one template: 0.36, multiTemplate: 0.37). Same holds for a couple more that I checked. 

-Is the configuration file correct? To my understanding, the first array is the template we use for the detection, and the second array contains origin IDs (these should be in the event.scml inventory) from which `scdetect-cc` will get the amplitudes for the regression. Right?

I think we can close this issue.

damb, 2022-02-16:
> This is the next step. I wanted to make sure it works and then we will check the results in detail! Keep in mind that the earthquake I use for detection is already small and it is normal to detect [...]

:+1:

damb, 2022-02-16:
> Is the configuration file correct? To my understanding, the first array is the template we use for the detection, and the second array contains origin IDs (these should be in the event.scml inventory) from which scdetect-cc will get the amplitudes for the regression. Right?

The formatting looks good. Though, I didn't run `scdetect-cc` with config since I didn't prepare a catalog (i.e. the `event.scml`) including the third-party origins. Also, I'd need the `templates.json` in order to check the if the `templates-family.json` is aligned.

The `"references"` JSON array includes references to either detector configurations (i.e. the relation is established by means of the `"detectorId"`) or third-party origins. Third-party origins add additional samples to the regression.

damb, 2022-02-16:
> I think we can close this issue.

:+1: