swiss-seismological-service/scdetect #70: Alternative magnitude implementation: amplitude ratio between template and detected event

damb, 2021-12-14:
### Discussed in https://github.com/damb/scdetect/discussions/55

<div type='discussions-op-text'>

<sup>Originally posted by **mmesim** November  4, 2021</sup>
Ideas for magnitude calculation based on what is already available and used in the literature (etc). 
- **Amplitude ratio between template and detected event**
e.g  https://www.nature.com/articles/ngeo697

[...]</div>

damb, 2022-01-05:
Both the amplitude and magnitude are going to be named `MRelative`.

damb, 2022-01-12:
@mmesim, for clarification: if a detector configuration includes a vertical instrument component, `MRelative` is going to be computed under consideration of that vertical component, too. Though, in any case the *reference* station magnitudes are `MLhc`/`MLh`. Do you agree with this?

mmesim, 2022-01-12:
Yes.

PR #74 (2022-01-13): Alternative magnitude implementation: MRelative
**Features and Changes**:
- Provide alternative *relative magnitude* implementation (in accordance with #70); subdevided into a `MRelative` amplitude and a `MRelative` magnitude.
- Implement both `AmplitudeProcessor` and `MagnitudeProcessor` factory facilities.
- Do some code refactoring.
- Allow users to provide sensor location specific configuration for `MRelative`.
- Provide a basic set of integration test covering both the `MRelative` amplitude and magnitude implementation

**Note**: This PR requires most recent upstream changes from SeisComP [... cut]
closed 2022-01-13 by damb