swiss-seismological-service/scdetect #81: Allow ringbuffer size to be configurable

damb, 2022-01-21:
The ringbuffer size used for waveform buffering is fixed at the time being. Allow it to be configurable.

Thanks to @luca-s for the hint.

damb, 2022-01-21:
See also https://www.seiscomp.de/doc/apps/scautopick.html#confval-ringBufferSize.

luca-s, 2022-01-21:
That is an useful option, however scautopick  `ringBufferSize` doesn't properly document what is useful for (one needs to look in the source code to understand that) and it doesn't explain the implication of selecting a certain value. I wish scautopick could figure it out by itself what is the optimal  `ringBufferSize` size, or, if that is not possible, provided some sensible guidelines on how to configure that buffer size

PR #84 (2022-02-04): Allow to configure the waveform buffer size
**Features and Changes**:
- Allow to configure the waveform buffer size (in accordance with `scautopick`'s `ringBufferSize` configuration parameter); currently `scdetect-cc` does not compute the buffer size by itself - this may be implemented in future.
- Note that in case the buffer size is too small amplitude calculation (and therefore magnitude calculation, too) fails.
- Correctly handle failing amplitude processors.

Closes #81.
closed 2022-02-04 by damb