swiss-seismological-service/scdetect #86: Tests for multi-stream detections

mmesim, 2022-02-09:
Provide tests for multi-stream [multi-station/ multi-channel] detections. Ideally, data similar to #62 and #66.

damb, 2022-02-15:
I assume you're using the same template waveform data as in #62, or rather #66, respectively. I'm asking since the data provided does not include the waveform data chunks required for template waveform creation.

damb, 2022-02-15:
Which of the detections do you get?

mmesim, 2022-02-15:
> I assume you're using the same template waveform data as in #62, or rather #66, respectively. I'm asking since the data provided does not include the waveform data chunks required for template waveform creation.

Yes. Sorry I forgot to add the templates to the mseed files :pensive:

damb, 2022-02-15:
> @damb Any clue? Can you re-produce the error?

Regarding the multi channel test I obtain the detections:
```
2019-11-05T04:01:11.880489 46.32480762 7.36023502 4.813476562 0.848927
2019-11-05T04:01:29.260489 46.32480762 7.36023502 4.813476562 0.534981
2019-11-05T04:01:40.125489 46.32480762 7.36023502 4.813476562 0.665105
```

Due to the `"initTime"` of `10 s` it is clear why we do not obtain the very first detection (i.e. as documented the detector is blind during the initialization time).

Though, it is interesting, why `scdetect-cc` does not emerge the very last detection:
```
2019-11-05T04:01:44.905489 46.32480762 7.36023502 4.813476562 0.562986
```

It is detected if adding 5 s more data (i.e. `"http://arclink.ethz.ch/fdsnws/dataselect/1/query?net=8D&sta=RAW2&cha=HHN,HHE&start=2019-11-05T04:01:00&end=2019-11-05T04:02:05"`). Actually, this requires a closer look, particularly, since the `8D.RAW2..` `Sg` arrival offset is ~ 3 s, only. With a template waveform length of 2.5 s the detection should be included. Currently, I'm not aware off additional latencies caused by the configuration. Maybe due to the filter applied, but this is just a guess.

mmesim, 2022-02-15:
OK. I couldn't detect these earthquakes because I think I was in the wrong branch. I changed VMs, working at the office now, and seems something broke or whatever. I'm fixing it and I'll update here. Seems that I was the problem.

mmesim, 2022-02-15:
Well, to wrap up. 

- I was in the wrong branch! :open_mouth: 
- I re-compiled everything and I confirm that I get 3 detections from the first test https://github.com/damb/scdetect/issues/86#issuecomment-1039339099 . 
Missing the first and last detection can be an issue but we can check that again later on the online implementation. Maybe wait a few seconds before run detections, or something like that. 
- For this test https://github.com/damb/scdetect/issues/86#issuecomment-1039353035 I get 1 detection. I took a longer waveform, 70s to 120s and still cannot find more detections. **BUT**, I did another test. 5 min long wvf, expected detections 
```
2019-11-05T04:30:30.010489Z  46.3248    7.3602    4.81 0.64 --detected
2019-11-05T04:30:43.940489Z  46.3248    7.3602    4.81 0.73 --detected
2019-11-05T04:32:29.820489Z  46.3248    7.3602    4.81 0.82 --detected
2019-11-05T04:33:17.385489Z  46.3248    7.3602    4.81 0.72 --missed
``` 

@damb What do you think? Is this something major or we can live with that for the time being? Do you think it is ok to use these tests and merge the feature to the master? If so, then I can work on the magnitudes.

damb, 2022-02-15:
> Missing the first and last detection can be an issue but we can check that again later on the online implementation.

As already mentioned, missing the first detection is due to the configuration you provided. The situation is completely different if the *last* detection is not detected. I assume, in this case there is still somewhere a bug.

I confirm the results for https://github.com/damb/scdetect/issues/86#issuecomment-1039353035. Again, it is the last detection which is not detected.

> @damb What do you think? Is this something major or we can live with that for the time being? Do you think it is ok to use these tests and merge the feature to the master? If so, then I can work on the magnitudes.

Well, the bug needs to be fixed. Therefore, I'm going to open a new issue. When fixing the bug, I'm going to add the tests from this issue, here. However, since the current implementation is already *kind of fixed* I'm going to merge #85  back into master, beforehand.

mmesim, 2022-02-15:
Great!

damb, 2022-02-15:
Superseded by #87.

closed 2022-02-15 by damb