swiss-seismological-service/scdetect #93: Magnitudes with nan

mmesim, 2022-02-23:
Sometimes magnitudes cannot be calculated and the scml file looks like

``` xml
      <stationMagnitude publicID="StationMagnitude/20220223105303.993555.18443">
        <magnitude>
          <value>nan</value>
        </magnitude>
        <type>MRelative</type>
```

When I convert the scml to xml (obspy friendly version) I receive 
```
ValueError: On StationMagnitude object: Value 'nan' for 'mag' is not a finite floating point value.
```

Perhaps that will cause problems in the future? I'm not sure about that, just a thought. One suggestion would be to add a crazy value like -99.99 for the magnitudes that cannot be calculated. I don't suggest -9 because if  `scdetecct-cc` is going to be used for labquakes then -9 is a possible magnitude. 

@damb thoughts?

damb, 2022-02-23:
@mmesim, thanks for spotting the issue.

Instead of adding a value like `-99.99`, I'm in favor of not adding a station magnitude at all. What do you think?

damb, 2022-02-23:
Also, could you provide your configuration in order to reproduce the issue. Thanks.

mmesim, 2022-02-23:
> @mmesim, thanks for spotting the issue.
> 
> Instead of adding a value like `-99.99`, I'm in favor of not adding a station magnitude at all. What do you think?

It is a good idea if it works in the end.

mmesim, 2022-02-23:
> Also, could you provide your configuration in order to reproduce the issue. Thanks.

of course.
[19.test.zip](https://github.com/damb/scdetect/files/8123920/19.test.zip)

damb, 2022-02-23:
Thanks @mmesim. I need to look into it and will come back.

closed 2022-02-25 by damb