swiss-seismological-service/scdetect #97: Multi-template detections & Magnitude calculations crushes

mmesim, 2022-03-09:
Well, 
I was trying to run `scdetect-cc` as :

- 10 detectors 
- 30 templates
- 1 station
- 3 channels

When running the same dataset without requiring magnitude calculation, I was able to get results, i.e 3330 origins -> 1109 unique events. 

With the magnitude on 

``` json

"createAmpliutdes": true
"createMagnitudes": true

``` 

I saw this error many times 

```
13:58:47 [error] [detector-02::8b18b121-6d0a-4a80-86f9-f8f8e84491aa::aa538897-997e-4219-971c-eb3a9e89a256] [8D.RAW2..HHN] Failed to feed data (tw.start=2019-11-05T01:14:32.455Z, tw.end=2019-11-05T01:14:34.355Z). Reason: status=101, status_value=0.000000 (failed to feed data)
```
At some point it crushed 

```
=================================================================
==3740==ERROR: AddressSanitizer: heap-use-after-free on address 0x6060011426e8 at pc 0x560006a50d55 bp 0x7ffc649ae000 sp 0x7ffc649adff0
READ of size 8 at 0x6060011426e8 thread T0
    #0 0x560006a50d54 in std::__shared_ptr<Seiscomp::detect::Application::DetectionItem, (__gnu_cxx::_Lock_policy)2>::get() const /usr/include/c++/9/bits/shared_ptr_base.h:1310
    #1 0x560006a3f00b in bool std::operator==<Seiscomp::detect::Application::DetectionItem, Seiscomp::detect::Application::DetectionItem>(std::shared_ptr<Seiscomp::detect::Application::DetectionItem> const&, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> const&) /usr/include/c++/9/bits/shared_ptr.h:384
    #2 0x560006a2019d in Seiscomp::detect::Application::removeDetection(std::shared_ptr<Seiscomp::detect::Application::DetectionItem> const&) **/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/app.cpp:1925
    #3 0x560006a1bd39 in Seiscomp::detect::Application::publishAndRemoveDetection(std::shared_ptr<Seiscomp::detect::Application::DetectionItem>&) **/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/app.cpp:1752
    #4 0x560006a00642 in Seiscomp::detect::Application::done() **/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/app.cpp:507
    #5 0x7f3e9432b564 in Seiscomp::System::Application::exec() **/work/projects/seiscomp/src/base/common/libs/seiscomp/system/application.cpp:1105
    #6 0x7f3e9432b451 in Seiscomp::System::Application::operator()() **/work/projects/seiscomp/src/base/common/libs/seiscomp/system/application.cpp:1079
    #7 0x560006c1133c in main **/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/main.cpp:11
    #8 0x7f3e92f9c0b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x240b2)
    #9 0x5600069880dd in _start (**/seiscomp/bin/scdetect-cc+0x3960dd)

0x6060011426e8 is located 40 bytes inside of 64-byte region [0x6060011426c0,0x606001142700)
freed by thread T0 here:
    #0 0x7f3e9581a8df in operator delete(void*) (/lib/x86_64-linux-gnu/libasan.so.5+0x1108df)
    #1 0x560006a88ac7 in __gnu_cxx::new_allocator<std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, true> >::deallocate(std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, true>*, unsigned long) /usr/include/c++/9/ext/new_allocator.h:128
    #2 0x560006a7f661 in std::allocator_traits<std::allocator<std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, true> > >::deallocate(std::allocator<std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, true> >&, std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, true>*, unsigned long) /usr/include/c++/9/bits/alloc_traits.h:470
    #3 0x560006a722f4 in std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, true> > >::_M_deallocate_node_ptr(std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, true>*) /usr/include/c++/9/bits/hashtable_policy.h:2113
    #4 0x560006a63511 in std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, true> > >::_M_deallocate_node(std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, true>*) /usr/include/c++/9/bits/hashtable_policy.h:2103
    #5 0x560006a715ec in std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> > >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, false> >::_M_erase(unsigned long, std::__detail::_Hash_node_base*, std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, true>*) (**/seiscomp/bin/scdetect-cc+0x47f5ec)
    #6 0x560006a61642 in std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> > >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, false> >::erase(std::__detail::_Node_const_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, false, true>) /usr/include/c++/9/bits/hashtable.h:1861
    #7 0x560006a50e5c in std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> > >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, false> >::erase(std::__detail::_Node_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<Seiscomp::detect::Application::DetectionItem> >, false, true>) /usr/include/c++/9/bits/hashtable.h:768
[... cut ...]
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==3740==ABORTING
```

Any clue? Is the VM or the code?

damb, 2022-03-09:
Thanks for the report.
> I saw this error many times
> ```
> 13:58:47 [error] [detector-02::8b18b121-6d0a-4a80-86f9-f8f8e84491aa::aa538897-997e-4219-971c-eb3a9e89a256] [8D.RAW2..HHN] Failed to feed data [...]

What's the configured `waveformBufferSize`? Is there data for the required waveform stream identifier (i.e. `8D.RAW2..HHN`) and time window available?

damb, 2022-03-09:
> Any clue? Is the VM or the code?

There is a memory issue during the application clean-up. Could you provide data including the configuration in order to reproduce the issue? Thanks.

mmesim, 2022-03-09:
> What's the configured `waveformBufferSize`? Is there data for the required waveform stream identifier (i.e. `8D.RAW2..HHN`) and time window available?

I didn't set any attribute `waveformBufferSize`, so I guess is the default (?).

mmesim, 2022-03-10:
I just found it in `scconfig` :open_mouth: 

It was 300 seconds. I think a value of 60 seconds would be ok, right?

damb, 2022-03-10:
> It was 300 seconds. I think a value of 60 seconds would be ok, right?

`scautopick`'s default is 300 s (see https://docs.gempa.de/seiscomp/current/apps/scautopick.html#confval-ringBufferSize). For the config you run this shouldn't be an issue (though, less is better in order to reduce `scdetect-cc`'s memory footprint). Note that the required data for amplitude calculation is still must be buffered. Else amplitude calculation (and therefore magnitude calculation, too) fails.

damb, 2022-03-14:
@mmesim, I had a closer look into the issue.

First of all, the memory issue should be fixed with #100.

Besides, I had a look into the issue causing the error log entry

```
13:58:47 [error] [detector-02::8b18b121-6d0a-4a80-86f9-f8f8e84491aa::aa538897-997e-4219-971c-eb3a9e89a256] [8D.RAW2..HHN] Failed to feed data (tw.start=2019-11-05T01:14:32.455Z, tw.end=2019-11-05T01:14:34.355Z). Reason: status=101, status_value=0.000000 (failed to feed data)
```

In order to understand what's going on behind the scenes it is important to remind that data required for amplitude calculation must be still buffered. Though, **if trigger facilities are enabled** the current detection is emitted (and therefore processed) only once the next detection is available. However, if the time passed by (i.e. the time between the current detection and the next one) was greater than the time specified by `--processing.waveformBufferSize` no buffered data is available anymore and thus amplitude calculation fails. As a result you see an error log entry as e.g. above. With #100 I handled this case more gracefully. Though, the fact that there is no amplitude calculated cannot be avoided. AFAIU, the issue can be partially avoided only if fetching historical data is implemented (e.g. similar to the approach by [`scamp`](https://docs.gempa.de/seiscomp/current/apps/scamp.html)).

PR #100 (2022-03-14): Fix issues reported by #97
**Bugfixes**:
- Fix memory issue when flushing pending detections

**Features and Changes**:
- Handle amplitude processor errors caused by missing buffered data more gracefully (i.e. report a warning instead of an error). Note that at the time being, fetching historical data is not implemented, yet.

Closes #97.
closed 2022-03-14 by damb