swiss-seismological-service/scdetect #98: Memory issue

mmesim, 2022-03-10:
I got this error message while executing another example. 

This time the amplitude and magnitude calculations were off. 

Specs:

24 hours of data
8 detectors
24 templates
1 station
3 channels

```
10:58:51 [debug] [detector-08] Start processing detection (time=2019-07-30T01:53:24.226292Z, associated_results=3) ...
10:58:51 [debug] Found 67 origins.
10:58:51 [debug] Leaving ::done
10:58:51 [info] Waiting for record thread
10:58:51 [debug] Unload plugin 'PostgreSQL database driver'

=================================================================
==19975==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 3840 byte(s) in 16 object(s) allocated from:
    #0 0x7fd3af9c5587 in operator new(unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cc:104
    #1 0x55f3e77b1956 in std::unique_ptr<Seiscomp::GenericRecord, std::default_delete<Seiscomp::GenericRecord> > Seiscomp::detect::util::make_unique<Seiscomp::GenericRecord, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Seiscomp::Core::Time, double>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Seiscomp::Core::Time&&, double&&) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/processing/detail/../../util/memory.h:16
    #2 0x55f3e77b0ba9 in Seiscomp::detect::processing::detail::InterpolateGaps::fillGap(Seiscomp::detect::processing::StreamState&, Seiscomp::Record const*, Seiscomp::Core::TimeSpan const&, double, unsigned long) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/processing/detail/gap_interpolate.cpp:77
    #3 0x55f3e77b0450 in Seiscomp::detect::processing::detail::InterpolateGaps::handleGap(Seiscomp::detect::processing::StreamState&, Seiscomp::Record const*, boost::intrusive_ptr<Seiscomp::NumericArray<double> >&) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/processing/detail/gap_interpolate.cpp:50
    #4 0x55f3e77b5287 in Seiscomp::detect::processing::WaveformProcessor::store(Seiscomp::Record const*) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/processing/waveform_processor.cpp:116
    #5 0x55f3e7736ce4 in Seiscomp::detect::detector::Detector::store(Seiscomp::Record const*) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/detector/detector.cpp:464
    #6 0x55f3e77b495c in Seiscomp::detect::processing::WaveformProcessor::feed(Seiscomp::Record const*) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/processing/waveform_processor.cpp:60
    #7 0x55f3e7596a84 in Seiscomp::detect::Application::handleRecord(Seiscomp::Record*) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/app.cpp:587
    #8 0x7fd3af5dfe60 in Seiscomp::Client::StreamApplication::dispatch(Seiscomp::Core::BaseObject*) /home/maria/work/projects/seiscomp/src/base/common/libs/seiscomp/client/streamapplication.cpp:151
    #9 0x55f3e7595ac1 in Seiscomp::detect::Application::dispatch(Seiscomp::Core::BaseObject*) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/app.cpp:533
    #10 0x7fd3af5b16a9 in Seiscomp::Client::Application::processEvent() /home/maria/work/projects/seiscomp/src/base/common/libs/seiscomp/client/application.cpp:1298
    #11 0x7fd3af5b1526 in Seiscomp::Client::Application::run() /home/maria/work/projects/seiscomp/src/base/common/libs/seiscomp/client/application.cpp:1274
    #12 0x7fd3af5dfd0e in Seiscomp::Client::StreamApplication::run() /home/maria/work/projects/seiscomp/src/base/common/libs/seiscomp/client/streamapplication.cpp:110
    #13 0x55f3e759517d in Seiscomp::detect::Application::run() /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/app.cpp:495
    #14 0x7fd3ae4d750d in Seiscomp::System::Application::exec() /home/maria/work/projects/seiscomp/src/base/common/libs/seiscomp/system/application.cpp:1099
    #15 0x7fd3ae4d7451 in Seiscomp::System::Application::operator()() /home/maria/work/projects/seiscomp/src/base/common/libs/seiscomp/system/application.cpp:1079
    #16 0x55f3e77a633c in main /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/main.cpp:11
    #17 0x7fd3ad1480b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x240b2)

Indirect leak of 24576 byte(s) in 16 object(s) allocated from:
    #0 0x7fd3af9c5587 in operator new(unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cc:104
    #1 0x55f3e756cfd8 in __gnu_cxx::new_allocator<double>::allocate(unsigned long, void const*) /usr/include/c++/9/ext/new_allocator.h:114
    #2 0x55f3e756c085 in std::allocator_traits<std::allocator<double> >::allocate(std::allocator<double>&, unsigned long) /usr/include/c++/9/bits/alloc_traits.h:444
    #3 0x55f3e756b2d3 in std::_Vector_base<double, std::allocator<double> >::_M_allocate(unsigned long) /usr/include/c++/9/bits/stl_vector.h:343
    #4 0x55f3e7569719 in void std::vector<double, std::allocator<double> >::_M_realloc_insert<double>(__gnu_cxx::__normal_iterator<double*, std::vector<double, std::allocator<double> > >, double&&) /usr/include/c++/9/bits/vector.tcc:440
    #5 0x55f3e75679c6 in void std::vector<double, std::allocator<double> >::emplace_back<double>(double&&) /usr/include/c++/9/bits/vector.tcc:121
    #6 0x55f3e7565d83 in std::vector<double, std::allocator<double> >::push_back(double&&) /usr/include/c++/9/bits/stl_vector.h:1201
    #7 0x55f3e776b09f in std::back_insert_iterator<std::vector<double, std::allocator<double> > >::operator=(double&&) /usr/include/c++/9/bits/stl_iterator.h:522
    #8 0x7fd3ae47f0f3 in std::back_insert_iterator<std::vector<double, std::allocator<double> > > std::transform<double const*, std::back_insert_iterator<std::vector<double, std::allocator<double> > >, convert<double, double> >(double const*, double const*, std::back_insert_iterator<std::vector<double, std::allocator<double> > >, convert<double, double>) (/home/maria/seiscomp/lib/libseiscomp_core.so.15+0xd4b0f3)
    #9 0x7fd3ae47dcb7 in void convertArray<std::vector<double, std::allocator<double> >, double>(std::vector<double, std::allocator<double> >&, int, double const*) (/home/maria/seiscomp/lib/libseiscomp_core.so.15+0xd49cb7)
    #10 0x7fd3ae47c5a3 in Seiscomp::ArrayFactory::Create(Seiscomp::Array::DataType, Seiscomp::Array::DataType, int, void const*) /home/maria/work/projects/seiscomp/src/base/common/libs/seiscomp/core/arrayfactory.cpp:209
    #11 0x7fd3ae4b4c46 in Seiscomp::GenericRecord::setData(int, void const*, Seiscomp::Array::DataType) /home/maria/work/projects/seiscomp/src/base/common/libs/seiscomp/core/genericrecord.cpp:152
    #12 0x55f3e77b0dad in Seiscomp::detect::processing::detail::InterpolateGaps::fillGap(Seiscomp::detect::processing::StreamState&, Seiscomp::Record const*, Seiscomp::Core::TimeSpan const&, double, unsigned long) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/processing/detail/gap_interpolate.cpp:88
    #13 0x55f3e77b0450 in Seiscomp::detect::processing::detail::InterpolateGaps::handleGap(Seiscomp::detect::processing::StreamState&, Seiscomp::Record const*, boost::intrusive_ptr<Seiscomp::NumericArray<double> >&) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/processing/detail/gap_interpolate.cpp:50
    #14 0x55f3e77b5287 in Seiscomp::detect::processing::WaveformProcessor::store(Seiscomp::Record const*) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/processing/waveform_processor.cpp:116
    #15 0x55f3e7736ce4 in Seiscomp::detect::detector::Detector::store(Seiscomp::Record const*) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/detector/detector.cpp:464
    #16 0x55f3e77b495c in Seiscomp::detect::processing::WaveformProcessor::feed(Seiscomp::Record const*) /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/processing/waveform_processor.cpp:60
[... cut ...]
    #15 0x7fd3ae4d750d in Seiscomp::System::Application::exec() /home/maria/work/projects/seiscomp/src/base/common/libs/seiscomp/system/application.cpp:1099
    #16 0x7fd3ae4d7451 in Seiscomp::System::Application::operator()() /home/maria/work/projects/seiscomp/src/base/common/libs/seiscomp/system/application.cpp:1079
    #17 0x55f3e77a633c in main /home/maria/work/projects/seiscomp/src/extras/scdetect/src/apps/cc/main.cpp:11
    #18 0x7fd3ad1480b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x240b2)

SUMMARY: AddressSanitizer: 29056 byte(s) leaked in 48 allocation(s).

damb, 2022-03-10:
Thanks for the report, @mmesim.

Apparently, there is a bug when interpolating gaps. Do you have the data/config in order to reproduce the issue? Thanks.

damb, 2022-03-10:
Thanks for the data. I can reproduce the bug. I'll come back once it is fixed.

PR #99 (2022-03-14): Fix gap interpolation
Closes #98.
closed 2022-03-14 by damb