swiss-seismological-service/scdetect #101: minimumArrivals

mmesim, 2022-03-21:
@damb Well, I shared with you the files. Here I'll report the problem. I run the shared example and set `minimuArrivals`: 10. However, when I run `scevent` with the results.scml I received the following message multiple times

```
rejecting automatic origin Origin/20220318161026.734837.31225 (phaseCount: 3 < 10)
```
or 
```
rejecting automatic origin Origin/20220318161026.734837.31225 (phaseCount: 6 < 10)
```

That probably means that there are not 10 picks available, which seems weird. 

I still do not if it `scdetect-cc` or `scevent` problem. 

I didn't include the results.scml to check whether you can fully reproduce the error.

damb, 2022-03-21:
@mmesim, thanks for reporting.

> That probably means that there are not 10 picks available, which seems weird.

Without having checked the reproducer's configuration yet, I guess that this might be an issue when trying to avoid to associate a single phase to be associated multiple times with a sensor location. This is required in certain configuration scenarios, e.g. given the following detector configuration:

```json
  {
    "detectorId": "detector-01",
    "createArrivals": true,
    "originId": "originId",
    "templatePhase": "Sg",
    "templateWaveformStart": -0.25,
    "templateWaveformEnd": 2,
    "streams": [
      {
        "templateId": "template-01",
        "waveformId": "CH.GRIMS..HHN"
      },
      {
        "templateId": "template-02",
        "waveformId": "CH.GRIMS..HHE"
      }
    ]
  }
```
I.e. in the example above, in order to avoid to associate an `Arrival` with phase `Sg`  twice (regarding the sensor location `CH.GRIMS..`) `scdetect-cc` drops one of the arrivals including the respective pick. This is not ideal, however, AFAIU, in the SeisComP datamodel the relationship `Pick <-> Origin` can be established only by means of an `Arrival`.

Do you have a better idea how to circumvent this issue?

mmesim, 2022-03-21:
What I do not understand is why I didn't face this problem with the other testing datasets. 

If what you mention above is the case, then it should have happened before. Still thinking about it...

damb, 2022-03-21:
> If what you mention above is the case, then it should have happened before. Still thinking about it...

- You mean, while using the `"minimumArrivals"` configration parameter, as well?
- Did you check `scevent` logs previously, too?

mmesim, 2022-03-21:
> You mean, while using the "minimumArrivals" configration parameter, as well?
Yes.

> Did you check scevent logs previously, too?
Yes

mmesim, 2022-03-22:
I just realized that there was a typo. Instead of writing `ELM0` I wrote `EML0`. :face_exhaling: 

Also, I do not know why `STLM2.HG?` is not in the inventory. Perhaps I did not include acc? I don't remember. 

To conclude, in the case that `scdetect-cc` will not be able to identify  10 phases, what will it do and how is this trackable/manageable in real time application?

damb, 2022-03-22:
> To conclude, in the case that `scdetect-cc` will not be able to identify 10 phases, what will it do and how is this trackable/manageable in real time application?

Warnings are already issued. Though, in case the `"minimumArrivals"` is configured `scdetect-cc` may not initialize the detector at all. What do you suggest?

mmesim, 2022-03-22:
Need to think about it.

closed 2022-03-24 by damb