swiss-seismological-service/scdetect #102: Allow to configure how to behave if detector initialization fails

damb, 2022-03-24:
Detector initialization may fail due to several reasons, e.g.:
- missing waveform data required for template waveform creation
- missing inventory metadata regarding a stream
- missing template event metadata
- ...

At the time being, `scdetect-cc`  implements the best-effort approach, i.e. the detector is initialized based on the information available. If information required is missing the corresponding stream is skipped, and the corresponding reason is logged. In order to skip initializing the detector entirely provide a module configuration parameter as an alternative approach.

Supersedes #101.

luca-s, 2023-02-27:
I don't believe `scdetect` should start for all the aforementioned cases. At least the template data and metadata must be fully available to start the processing. Then `scdetect` can be flexible with regard to real-time data, but allowing it to work without a proper initialization feels dangerous because a user would look at the logs only when something wrong happens (e.g. `scdetect` doesn't start), but if `scdetect` starts then a user would never realize that the template data is missing.

mmesim, 2023-02-27:
I agree @luca-s.

luca-s, 2023-03-01:
It looks like @damb made a beautiful job as usual and the behaviour is totally configurable via code already. 

Currently it is set to be very flexible and for the reason explained above I would make scdetect strict and fails if the templates cannot be properly loaded. It's just a matter of chaning few variables in the code, thanks a lot @damb!

I believe @damb wanted to make this configuration visible at user level, but unless someone show a valid use case for this, I would keep the configuration hidden.

luca-s, 2023-03-01:
Closed by [this ](https://github.com/swiss-seismological-service/scdetect/commit/c4970c8ca0a1f488ac43e8c2cc1ee6c1def9ae25) closes

closed 2023-03-01 by luca-s