swiss-seismological-service/scdetect #103: Network Magnitudes

mmesim, 2022-03-25:
It would be useful to get  network magnitudes for each detection. 
Then during the post-processing with `scolv` we can screen magnitudes easily.

luca-s, 2022-05-06:
If you are talking only about unassociated origins your understanding is correct, you won't be able to see the magnitude. That is because each origin might have multiple magnitudes and scevent task (one of many) is to select the preferred magnitude among all the origins and among all the magnitudes. You see that selected/preferred magnitude in the event list tab.However if an origin is not associated, it means scevent was not run or it was not able to create an event for that origin (because it doesn't satisfy the parameters configured for scevent). Without an event, there is no magnitude visualized. However you should be able to view all origin available magnitudes in the Magnitudes tab

damb, 2022-05-09:
> However you should be able to view all origin available magnitudes in the Magnitudes tab.

@mmesim, can you confirm that the StationMagnitudes are displayed after adding e.g. `MRelative` to the list of `visibleMagnitudes`?

mmesim, 2022-05-09:
I cannot see StationMAgnitudes.

mmesim, 2022-05-10:
I had a short conversation with @luca-s yesterday and it is more clear to me what happened. 

In brief, and @luca-s please correct me if I'm wrong, 

- **Unassociated origins**: they do not have magnitudes because `scevent` did not run, so we do not have preferred solution or preferred magnitude.
- **Seiscomp Events** (e.g. CH-quakes): an event has been created, `scevent` selected the prefered origin and associated multiple origins. Magnitudes are shown on the event list and also station magnitudes on the Magnitude tab. 
- **`scdetect-cc` events**:  There is no magnitude shown on the event list, because we do not calculate Network Magnitudes. Although there are station magnitudes, i.e. when you export the event from the database, these are not displayed on the magnitude tab. That's because there is no event magnitude and thus the station magnitudes are not displayed. 

To conclude, it is crucial to get network magnitudes.

luca-s, 2022-05-10:
@damb basically in the scdetect-cc origin we found the `stationMagnitude` object but the `magnitude` one is missing. Please have a look how a `magnitude` looks like:

[image]

damb, 2022-05-10:
@mmesim, which approach do you suggest to compute the [NetworkMagnitude](https://docs.gempa.de/seiscomp/current/apps/scmag.html#scmag-networkm)? Do you require it to be configurable?

@luca-s  thanks.

mmesim, 2022-05-10:
I think the median of the station magnitudes will do the trick. 

No

PR #106 (2022-05-16): Compute network magnitudes
Compute [network magnitudes](https://docs.gempa.de/seiscomp/current/apps/scmag.html#network-magnitudes) as the median of the corresponding station magnitudes.

Closes #103.
closed 2022-05-16 by damb