swiss-seismological-service/scdetect #112: error: ‘ofstream’ is not a member of ‘fs’; did you mean ‘std::ofstream’?

morealaz, 2022-07-05:
Hi,
I got this error when trying to build seiscomp with scdetect:
```
seiscomp/src/extras/scdetect/src/apps/cc/test/integration.cpp:255:9: error: ‘ofstream’ is not a member of ‘fs’; did you mean ‘std::ofstream’?
  255 |     fs::ofstream{pathConfig};
      |         ^~~~~~~~
In file included from /usr/include/boost/smart_ptr/intrusive_ptr.hpp:28,
                 from /usr/include/boost/intrusive_ptr.hpp:16,
                 from seiscomp/src/extras/scdetect/../../base/common/libs/seiscomp/core/defs.h:24:
/usr/include/c++/12.1.0/iosfwd:165:41: note: ‘std::ofstream’ declared here
  165 |   typedef basic_ofstream<char>          ofstream;
      |                                         ^~~~~~~~
```

damb, 2022-07-05:
Hi,

thanks for reporting.

- Which compiler are you using? Which version?
- Did you install the *boost libraries*, particularly `boost::filesystem`? I assume, yes. Which version? Most probably it's because of `boost::filesystem` stopped to include the `fstream.hpp` header file automatically.

damb, 2022-07-05:
Could you check, whether #113 fixes the compilation issue on your side? Thanks.

morealaz, 2022-07-05:
Hi Daniel,
>     * Which compiler are you using? Which version?
GCC 12.0.1
>     * Did you install the _boost libraries_, particularly `boost::filesystem`? I assume, yes. Which version? 
Yes, boost 1.79.0

BTW I use Archlinux

morealaz, 2022-07-05:
> Could you check, whether #113 fixes the compilation issue on your side? Thanks.

Yes, It fixed that error but I got following errors from linker:
```
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::open()':
sqlite.cpp:(.text+0x354): undefined reference to `sqlite3_open'
/usr/bin/ld: sqlite.cpp:(.text+0x413): undefined reference to `sqlite3_close'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::disconnect()':
sqlite.cpp:(.text+0x4f1): undefined reference to `sqlite3_close'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::execute(char const*)':
sqlite.cpp:(.text+0x63e): undefined reference to `sqlite3_exec'
/usr/bin/ld: sqlite.cpp:(.text+0x6fb): undefined reference to `sqlite3_free'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::beginQuery(char const*)':
sqlite.cpp:(.text+0x85b): undefined reference to `sqlite3_prepare'
/usr/bin/ld: sqlite.cpp:(.text+0x895): undefined reference to `sqlite3_column_count'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::endQuery()':
sqlite.cpp:(.text+0x8e9): undefined reference to `sqlite3_finalize'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::lastInsertId(char const*)':
sqlite.cpp:(.text+0x93f): undefined reference to `sqlite3_last_insert_rowid'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::numberOfAffectedRows()':
sqlite.cpp:(.text+0x97b): undefined reference to `sqlite3_changes'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::fetchRow()':
sqlite.cpp:(.text+0x9b3): undefined reference to `sqlite3_step'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::findColumn(char const*)':
sqlite.cpp:(.text+0x9ed): undefined reference to `sqlite3_column_name'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::getRowFieldName(int)':
sqlite.cpp:(.text+0xa63): undefined reference to `sqlite3_column_name'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::getRowField(int)':
sqlite.cpp:(.text+0xa8d): undefined reference to `sqlite3_column_blob'
/usr/bin/ld: CMakeFiles/test_scdetect_cc_integration.dir/__/detail/sqlite.cpp.o: in function `Seiscomp::detect::detail::SQLiteDatabase::getRowFieldSize(int)':
sqlite.cpp:(.text+0xab7): undefined reference to `sqlite3_column_bytes'
collect2: error: ld returned 1 exit status
make[2]: *** [src/extras/scdetect/src/apps/cc/test/CMakeFiles/test_scdetect_cc_integration.dir/build.make:984: bin/test_scdetect_cc_integration] Error 1
make[1]: *** [CMakeFiles/Makefile2:10329: src/extras/scdetect/src/apps/cc/test/CMakeFiles/test_scdetect_cc_integration.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
```
I use sqlite 3.39.0