swiss-seismological-service/scdetect #117: ctest fails partially after compiling with scdetect

funkes, 2022-07-07:
After compiling SeisComP with scdetect-cc <make test> runs to error messages comparable to #114 with the same 7 failed tests:

```
sysop@sxresearch:/var/opt/seiscomp/build$ 
sysop@sxresearch:/var/opt/seiscomp/build$ ctest --rerun-failed 
Test project /var/opt/seiscomp/build
    Start 21: static
1/7 Test #21: static ...........................***Failed    1.71 sec
    Start 22: event
2/7 Test #22: event ............................***Failed    2.49 sec
    Start 23: station
3/7 Test #23: station ..........................***Failed    2.83 sec
    Start 24: station-conditional
4/7 Test #24: station-conditional ..............***Failed    3.11 sec
    Start 25: dataselect
5/7 Test #25: dataselect .......................***Failed    2.45 sec
    Start 26: availability
6/7 Test #26: availability .....................***Failed    2.45 sec
    Start 31: test_scdetect_cc_integration
7/7 Test #31: test_scdetect_cc_integration .....***Failed    2.45 sec

0% tests passed, 7 tests failed out of 7

Total Test time (real) =  17.54 sec

The following tests FAILED:
     21 - static (Failed)
     22 - event (Failed)
     23 - station (Failed)
     24 - station-conditional (Failed)
     25 - dataselect (Failed)
     26 - availability (Failed)
     31 - test_scdetect_cc_integration (Failed)
Errors while running CTest
Output from these tests are in: /var/opt/seiscomp/build/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
sysop@sxresearch:/var/opt/seiscomp/build$ 
```

After cleaning up the build directory I tried to enabling SQLite support globally (Following #114): I changed SQLite support to ON:

```
sysop@sxresearch:/var/opt/seiscomp/build$ grep "SC_TRUNK_DB_SQLITE3:" CMakeCache.txt 
SC_TRUNK_DB_SQLITE3:BOOL=ON
sysop@sxresearch:/var/opt/seiscomp/build$ 
```

But the errors persist. An isolated test of "static" brings a hint to the mdoule mseedlite:

```
sysop@sxresearch:/var/opt/seiscomp/build$ ctest --output-on-failure -R static
Test project /var/opt/seiscomp/build
    Start 21: static
1/1 Test #21: static ...........................***Failed    1.74 sec
Traceback (most recent call last):
  File "/var/opt/seiscomp/src/base/main/apps/fdsnws/test/test-static.py", line 15, in <module>
    from seiscomp.fdsnws import event, station, dataselect, availability
  File "/var/opt/seiscomp/build/lib/python/seiscomp/fdsnws/dataselect.py", line 34, in <module>
    from seiscomp import mseedlite
ImportError: cannot import name 'mseedlite' from 'seiscomp' (/var/opt/seiscomp/build/lib/python/seiscomp/__init__.py)

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   1.78 sec

The following tests FAILED:
	 21 - static (Failed)
Errors while running CTest
sysop@sxresearch:/var/opt/seiscomp/build$ 
```

I'm using gcc version 11.2.0 on Ubuntu 22.04. I checked out the sources today from github follwing https://github.com/swiss-seismological-service/scdetect.

I'm no sure, if my topic ist better related to general SeisComP stuff or to the scdetect-cc project.

damb, 2022-07-07:
Thanks for reporting.

```
     21 - static (Failed)
     22 - event (Failed)
     23 - station (Failed)
     24 - station-conditional (Failed)
     25 - dataselect (Failed)
     26 - availability (Failed)
```

are related to SeisComP's `fdsnws` module. This is unrelated to `scdetect-cc`. (Most probably, these tests fail due to not enabling global SQLite support successfully.)

Let's focus towards running the `scdetect-cc` related tests:

- Can you confirm that `libsqlite3` is available on your system?
- Did you compile successfully (also after enabling SQLite support)? Any errors? Strictly speaking, `scdetect-cc` won't require you to enable SQLite support globally.
- Did you install SeisComP? Particularly, is `$SEISCOMP_ROOT/share/db/sqlite3.sql` available?

funkes, 2022-07-08:
Now the problem is solved: Calling "make test" *after* "make install" brings the last test "31 - test_scdetect_cc_integration" to a successful end. Probably it was my mistaken to call  "make test" *before* "make install", but all the other tests ended successfully in this order too (after compiling the whole SeisComP packages).

damb, 2022-07-08:
@funkes, perfect!

> Probably it was my mistaken to call "make test" before "make install", but all the other tests ended successfully in this order too (after compiling the whole SeisComP packages).

Actually, it is a bit fuzzy to have the requirement to first install SeisComP in order to run `scdetect-cc` related tests. Though, at the time being I don't come up with a better idea of how to improve the situation.

closed 2022-07-08 by damb