swiss-seismological-service/scdetect #118: MRelative: Use Network Magnitude (event magnitude) instead of Station Magnitude

mmesim, 2022-07-13:
When we compared MRelative magnitudes from `scdetec-cc` with other methods we observed that magnitudes are systematically overestimated. The problem was that we used StationMAgnitudes and the nearest to the source station. Recently SED introduced MLhc, which, if applied, in principle could correct this issue. 

Issue 01: there is no correction for events prior to 2021 (in the database) and it is not clear when this would be done. 
Issue 02: What if we deploy a station right after a Mainshock and there is no station specific correction? 
Issue 03: MLhc is introduced at SED but it is network dependent and perhaps other networks do not use similar station corrections. 

I think it should be good to use the network magnitude instead of station magnitude when computing MRealtive. 

@damb Is it doable?

damb, 2022-07-13:
I'm not sure whether I fully understand the issue raised.

> The problem was that we used StationMagnitudes and the nearest to the source station. 

It is correct that `scdetect-cc` uses StationMagnitudes. Whether it is the nearest to the source station depends on the detector configuration. I.e. `scdetect-cc` computes a StationMagnitude for each sensor location listed in the detector configuration (taking into account the corresponding StationMagnitudes from the template origin). StationMagnitudes of type `MLhc` are preferred of StationMagnitudes of type `MLh`. Finally, all StationMagnitudes are reduced to a single (Network)Magnitude.

So I see that under certain circumstances it might be preferable to take the (Network)Magnitude into account. For instance, in case the template origin is associated with a N sensor locations which contributed to the final (Network)Magnitude. However, due to e.g. performance reasons a `scdetect-cc` detector was set up using < N sensor locations. In this case I would expect the final `MRelative` magnitude to be biased.

damb, 2022-07-13:
So from your description, ideally a user of `scdetect-cc` could configure whether to use the StationMagnitude or the template origin's (Network)Magnitude when it comes to computing `MRelative`. Would this help?

mmesim, 2022-07-13:
I think the ideal would be to calculate Mrelative using only the network magnitude. 

However, which is the most feasible and faster solution? To have the option to choose between station and network magnitude or use by default only the network magnitude?

damb, 2022-07-13:
I'd opt for configurable solution with the NetworkMagnitude as the default.

mmesim, 2022-07-13:
Agreed.

damb, 2022-07-18:
@mmesim, a question regarding the NetworkMagnitude to be used:

In the [SeisComP datamodel](https://www.seiscomp.de/doc/base/api-python.html#module-seiscomp.datamodel) `DataModel::Magnitude` is a child object of `DataModel::Origin`. The multiplicity is `N<->1` i.e. multiple NetworkMagnitudes may be associated with a single origin. However, the *preferred magnitude* is defined by `DataModel::Event`. When it comes to selecting the NetworkMagnitude, using the event's preferred magnitude to me seems not intuitive, since this particular origin may be unrelated to the origin which is actually used for detector configuration. So, which *NetworkMagnitude* should be used when estimating `MRelative`?

damb, 2022-07-18:
As a nother side-effect, using the `DataModel::Event`'s *preferred magnitude* would involve that the `DataModel::EventParameters` must include
- the `DataModel::Event` the template origin belongs to, and
- the  `DataModel::Origin` the preferred magnitude belongs to.

This is definitely a disadvantage, compared to the current implementation, which is more intuitive when it comes to EventParameters configuration.

mmesim, 2022-07-21:
> @mmesim, a question regarding the NetworkMagnitude to be used:

> In the [SeisComP datamodel](https://www.seiscomp.de/doc/base/api-python.html#module-seiscomp.datamodel) >DataModel::Magnitude is a child object of DataModel::Origin. The multiplicity is N<->1 i.e. [...]

I understand that might be confusing, however, the preferred manual solution should contain the "preferred " magnitude for the given seismic network. Another issue that I see is that in case a network doesn't use MLh or MLhc, as we do. Then what magnitude is going to be selected as the reference magnitude? If we apply `scdetect-cc` internally then by default the preferred magnitude would be MLhc (or MLh for earlier earthquakes). Does this answer your question?

mmesim, 2022-07-21:
> As a nother side-effect, using the `DataModel::Event`'s _preferred magnitude_ would involve that the `DataModel::EventParameters` must include
>
> * the `DataModel::Event` the template origin belongs [...]

Does this mean that we have to include the EventID in the template configuration instead of the OriginID?

mmesim, 2022-07-21:
So, do you want to merge it or not?

damb, 2022-07-21:
Actually I'm not sure, yet.

Besides, I'd like to avoid some user specific *hacks* in the codebase.

mmesim, 2022-07-24:
@damb
Ok, I had an epiphany! 💡 One way to decide whether the Network or Station Magnitudes is the optimal solution is to compute for the same dataset magnitudes with the two different methods. I already have results from the Station Magnitude approach (master branch). I can re-run the dataset using the [feature/iss-118](https://github.com/swiss-seismological-service/scdetect/tree/feature/iss-118) and compare (i) the Frequency Magnitude distribution, (ii) and the magnitudes for the common detections (SED catalog and `scdetect-cc`). I hope I can do it by Tuesday or Wednesday.

mmesim, 2022-08-15:
@damb Question
I re-compiled `scdetect-cc` under  https://github.com/swiss-seismological-service/scdetect/tree/feature/iss-118 but I cannot get Magnitudes. 

``` bash
11:31:23 [warning] [detector-08::95d7132b-770a-4be8-94bd-209a648b60af] Failed to create station magnitude: failed to create magnitude processor (failed to configure template magnitude)
11:31:23 [debug] [CH.EMING..HHE] Removing time window processor: id=detector-08::2dd9a59
```

Thoughts?

damb, 2022-08-16:
Magnitudes of type `MRelative` require a template magnitude to be configured. However, as the log entry states, creating the magnitude processor failed (due to a missing template magnitude). As a consequence, the station magnitude could not be computed.

I guess you wanted to compute `MRelative` based on the event's (preferred) NetworkMagnitude. In this case you need the event including the origin containing the (preferred) NetworkMagnitude. (Though, without your configuration it is just a guess.)

Please, also make sure that you understand the [SeisComP datamodel](https://www.seiscomp.de/doc/base/api-python.html) and read and understand my comment from https://github.com/swiss-seismological-service/scdetect/issues/118#issuecomment-1191640053.

mmesim, 2022-08-16:
Yes, it work when I used an input file that contained the event and the preferred origin. Thanks a lot for the input.

damb, 2022-08-16:
> Yes, it work when I used an input file that contained the event and the preferred origin. Thanks a lot for the input.

Note that it is not the preferred origin which is taken into account. Instead, it is the preferred magnitude (which may be unrelated to the preferred origin). So in order to make things work, the user is required to provide
- the event
- the template origin
- the origin which contains the event's preferred magnitude.

mmesim, 2023-01-05:
@luca-s You can check here.