swiss-seismological-service/scdetect #123: typo in scdetect-cc for Dummies -- screen results --record-url?

funkes, 2022-08-03:
To show seismograms by scolv (topic scdetect-cc for Dummies -- Offline Playback -- examples ex-02 & ex-03 -- screen results) I changed the option "--record-url" of scolv from

"fdsnws://eida-federator.ethz.ch/fdsnws/dataselect/1/query"

to

"fdsnws://eida.ethz.ch".

Probably the very useful documentation contains a typo?

damb, 2022-08-07:
Thanks for reporting.

Technically, the [`eidaws-federator`](http://eida-federator.ethz.ch/) URL should work, as well. The required MSEED data shouldn't have restricted access permissions and therefore should be accessible.

Though, for the sake of consistency we may switch to ETHZ's `fdsnws-dataselect` endpoint URL. @mmesim, any strong opinions against switching?

mmesim, 2022-08-08:
> Though, for the sake of consistency we may switch to ETHZ's `fdsnws-dataselect` endpoint URL. @mmesim, any strong opinions against switching?

No, please go ahead. 
Thanks @funkes for reporting.

damb, 2022-08-08:
@funkes, could you please provide a PR? Thanks.

funkes, 2022-08-08:
See branch "patch-1"; hopefully it meets your suggestions?!

Sorry for my hesitance, but it was my first PR on GIT. So I'm not sure, whether I've found a proper way.

Are explicite hints to the related issues (#122, #123) useful?

closed 2022-08-08 by damb