swiss-seismological-service/scdetect #127: Describe how to display detection score in the `scolv` Events tab

damb, 2022-08-17:
Although, displaying a custom column in `scolv`'s *Events tab* is already described in the [`scolv` documentation](https://www.seiscomp.de/doc/apps/scolv.html#id4) it would be good to add a view sentences to the documentation how to display the detection score within an additional column. Particularly for less experienced users this would be very useful.

damb, 2022-08-17:
@mmesim , AFAIK you have a working configuration already. Could you do this? Thanks.

mmesim, 2022-09-07:
Well, I thought a bit about it and I came to the conclusion that this is a `SeisComP` related issue. The documentation on the `SeisComP` page is already very detailed. I'll add here a detail comment so in the future the `scdetect-cc` users will find an answer in this closed issue. 

`SeisComP` allows the user to add a custom column, for example in `scolv`, to show origin or event comments. In the case of `scdetect-cc` in the comments the user can find the `scdetectDetectorID`, the earthquake template that detect the new earthquake, and the `scdetectResultCCC`,  the cross correlation coefficient between the template and the new detection. 

To add a custom column please go here -->https://www.seiscomp.de/doc/apps/global_gui.html#confval-eventlist.visibleColumns and then edit either the `scolv.cfg` file or the `scconfig` GUI. 

**Please note**
The user can add **only one** comment, and thus **only one** column. 

For more columns, and thus more comments, please choose the `External Scripts` and the `Custom Information`. 

One example is here -- > https://www.seiscomp.de/doc/apps/scolv.html#custom-information-in-tabs

and continues here -->  https://www.seiscomp.de/doc/apps/scolv.html#sec-scolv-parameters-events 

**Note**
From the `SeisComP` page 

> Adding external scripts to the event list is the most expensive part. Whenever the event list is reloaded, it >will start to run the scripts on all origins in the background. It can take a while until the complete list has >been processed. A progress indicator is plotted in the middle of the event list while the background >processes are running.

damb, 2022-09-08:
Although, I agree that this issue is kind of SeisComP related, I'd have added a description (referencing the corresponding section of the SeisComP documentation) of how to display the detection score to the SCDetect documentation. This is for several reasons:
- First of all, for the user it is not clear that `scdetect-cc` attaches the detection score by means of a `DataModel::Comment` to the detection (i.e. the `DataModel::Origin`). Also, up to now from the user documentation, it is not clear which comment identifier is used (i.e. `scdetectResultCCC`).
- Second, because of improving and simplifying maintenance. What if e.g. the detection score comment identifier changes? Will somebody update the comment dropped within this issue? I guess rather not. On the other hand, it is highly probable that the SCDetect documentation is updated.
- Keeping things together, and avoiding parts of the documentation being scattered around, while referencing the SeisComP documentation makes it clear and concise for users to customize the configuration of other SeisComP modules while using `scdetect-cc`.

Note that the SCDetect documentation references the SeisComP documentation by means of the [intersphinx extension](https://www.sphinx-doc.org/en/master/usage/extensions/intersphinx.html).

mmesim, 2022-09-09:
You convinced me. :roll_eyes: :woman_facepalming:

closed 2023-08-09 by luca-s