swiss-seismological-service/scdetect #128: No detection for Offline Playback - Example 3

damb, 2022-08-22:
The template configuration of the [third offline playback example](https://scdetect.readthedocs.io/en/stable/base/SCDetect-for-Dummies.html#example-multi-detector-multi-station-multi-channel-detection) ([sources](https://github.com/swiss-seismological-service/scdetect/blob/3372889183d995f408a601245ba910ac4c8b079d/doc/base/Offline-Playback.rst)) contains overall three detector configurations. Though, when following the tutorial and running `scdetect-cc` there are no detections for the detector with `"detectorId"` `"detector-03"`. Is this on purpose, @mmesim?

mmesim, 2022-08-22:
> Is this on purpose, @mmesim?

No. I did not notice that.

damb, 2022-08-23:
Technically, it is fine if there is no detection. Usually, users don't know ahead of time whether there are detections for a specific template or not.

Anyway, providing the expected results (e.g. a table including the detection origin times and the detector identifiers) as part of the tutorial would IMO be of great value.

closed 2022-08-30 by damb