swiss-seismological-service/scdetect #135: Pick author not assigned

mmesim, 2022-12-21:
It seems that when a new pick is detected there is no author assigned 

[image]

@luca-s This might need to be fixed to be consistent with other pickers, e.g. scrtdd. Though it is not urgent.

luca-s, 2022-12-21:
I had a look at the code and it seems that the author is set, so I need some more information to understand what is going on. Can you explain better what is your scenario? Also it would be very useful if you can dump the detected origin with `scxmldump` so that I can have a better look at the creation information of the pick.

luca-s, 2022-12-22:
I fixed the issue on branch [fix-pick-author](https://github.com/swiss-seismological-service/scdetect/tree/fix-pick-author). IF you can give it a try I will then merge it to the `master` branch.

mmesim, 2022-12-23:
Thanks @luca-s! It works! Go ahead with the merge and close this issue. :fireworks:

closed 2022-12-23 by mmesim