swiss-seismological-service/scdetect #137: Detected picks have truncated channel codes (e.g. HH istead of HHZ)

luca-s, 2023-02-27:
Original pick:
```
    <pick publicID="smi:....">
      <waveformID networkCode="8R" stationCode="V0117" channelCode="JDD"/>
    </pick>
```

Detected pick (see the difference `JDD` vs `JD`):
```
    <pick publicID="smi:....">
      <waveformID networkCode="8R" stationCode="V0117" channelCode="JD"/>
    </pick>
```

This behaviour is on purpose, see [here](https://github.com/swiss-seismological-service/scdetect/blob/c3cccf8bbf5dca40215fb3a1c1dd5f0a25f5b482/src/apps/cc/app.cpp#L825) and it is implemented only when a pick is detected. When a pick is derived from the template then the 3 characters of the channel code are fully used.

@mmesim do you know why?

luca-s, 2023-02-27:
Ok I got it. This behaviour is required to avoid the creation of too many picks for a single detection (one per component). If a detector has 3 components configured for a single station, then the detection would have 3 picks without the current `scdetect` behaviour.  Since only one pick is reported for 3 components then the component part of the channel code is droppped. Makes sense. However in my case I only have one component per sensor, so I might need to improve the current behaviour.

mmesim, 2023-02-27:
Good.

luca-s, 2023-03-01:
I am closing this since I am not sure it is an important thing to change

closed 2023-03-01 by luca-s