swiss-seismological-service/scdetect #140: Cannot create templates for phases that are not used

mmesim, 2023-03-09:
It seems that when a phase pick is disable `scdetect-cc` cannot create a template. This is ok, when the phase is discarded due to large rms, for example. 

However we figured out that sometimes phases are not selected by `SeisComP` for other reasons, e.g. external picks imported in the database. 

@speedshi, @luca-s will fix it. Thanks for reporting!

luca-s, 2023-03-09:
I am tempted to change the code so that a pick/arrival is used independently of their enable/disable state. However there might be one specific use case for which @damb coded the current behaviour. It could be that for a specific phase, e.g. P, Pg, Pn etc, there are multiple arrivals, but only one is enabled. With the current code only the enabled one would be selected. So I need to think carefully before changing the code.

mmesim, 2023-03-10:
> It could be that for a specific phase, e.g. P, Pg, Pn etc, there are multiple arrivals, but only one is enabled.

From my experience with picking using `SeisComP` I'm almost certain that the user can pick multiple phases on a stream but only one each time, i.e. one Pg, one Sg, etc.. You cannot have two Pg for example. At least I haven't seen that, but I can test it. 

So, maybe the reason is elsewhere. Check also here https://github.com/swiss-seismological-service/scdetect/issues/21

luca-s, 2023-03-10:
So it turned out that the code makes use of the arrival weights, which are available only if a pick/arrival is enabled. So I cannot make the simple change of not considering the arrival status, because other part of the code would fail. To properly address this scenario I should also create a new configuration option that define the default pick/arrival weight in case that is not available.

luca-s, 2023-03-10:
Since this is a very remote use case, I don't plan to make any change to the code. However I leave here a workaround for the unlucky users who stumble across this issue:
- dump the origins you want to use to file e.g. `scxmldump -fPAMFJ -o origin1,origin2,originX -o templates.scml`
- edit the templates.scml file  by adding the missing info that `scdetect` wants: 
```
<arrival>
   ...
   <timeUsed>true</timeUsed>
   <weight>1</weight>
</arrival> 
```
- make use of `scdetect.eventDB` option to use the templates.scml file for your events