swiss-seismological-service/scdetect #144: Compile Source Code in SeisComPv6

akasa-denada, 2024-04-28:
Currently, I want to compile scdetect but I got error message like this. Can you help me?

[image]

luca-s, 2024-04-29:
Thanks for reporting the issue. We haven't switched to SeisComP 6 yet so we haven't noticed this compilation issue yet. Let me fix this and I will be back to you.

akasa-denada, 2024-04-29:
Thank you, Luca. I will wait for it.

luca-s, 2024-04-29:
If I compile the v6 branch the code compiles fine. Are you compiling the master branch of seiscomp @akasa-denada ?

akasa-denada, 2024-04-29:
Yes, I compile with seiscomp v6.3.1

luca-s, 2024-04-29:
>  The folder scdetect put in src/extra right?

Yes

> Yes, I compile with seiscomp v6.3.1.

So you mean "No", you don't compile the master branch. Because the seiscomp v6.3.1 is in the branch v6 not the master. could you confirm please?

akasa-denada, 2024-04-29:
Yes, I mean compile with the master branch

luca-s, 2024-04-29:
Ok now i understand. I tried compiling the master branch of seiscomp and I get the same error as you. 

However, unless you have a valid reason for it, you should not compiling the master branch. Each seiscomp version has a dedicated git branch (v4, v5, v6 ..) and a specific git tag for each released version (5.5.9, 6.0.0, 6.0.1, ...). If you intend to compile seiscomp 6 you should use branch v6. The master branch is a development branch that contains breaking changes and changes intended for the next major release of seiscomp (e.g. seiscomp 7). This is why you see this compilation error. If you compile the v6 branch all works fine and you are also guaranteed that the code is compatible with seiscomp 6 (e.g. database schema, but also API in case you install proprietary modules like scanloc).

So my suggestion is to checkout branch v6 and recompile.  That said, I need to fix the compilation issue anyway, because the master branch will become a new seiscomp version eventually. So I will provide a commit anyway.

Please let me know if my comment makes sense or if I was not clear with my explanation.

akasa-denada, 2024-04-29:
> Ok now i understand. I tried compiling the master branch of seiscomp and I get the same error as you.
> 
> However, unless you have a valid reason for it, you should not compile the master branch. Each seiscomp version has a dedicated git branch (v4, v5, v6 ..) and a specific git tag for each released version (5.5.9, 6.0.0, 6.0.1, ...). If you intend to compile seiscomp 6 you should use branch v6. The master branch is a development branch that contains breaking changes and changes intended for the next major release of seiscomp (e.g. seiscomp 7). This is why you see this compilation error. If you compile the v6 branch all works fine and you are also guaranteed that the code is compatible with seiscomp 6 (e.g. database schema, but also API in case you install proprietary modules like scanloc).
> 
> So my suggestion is to checkout branch v6 and recompile. That said, I need to fix the compilation issue anyway, because the master branch will become a new seiscomp version eventually. So I will provide a commit anyway.
> 
> Please let me know if my comment makes sense or if my explanation is not clear.

Thank you, Luca. I just tried it and it worked.

luca-s, 2024-04-29:
All good then. I am closing this.